Spirik/GEM #2: License not suitable for code?

matthijskooijman, 2018-01-08:
Looks like nice library, well-designed and clean code!

However, you've selected the CC license for it. That license is not intended to apply to code. I believe it will work, but there are other more specific licenses, such as the GPL (which should be similar in terms with CC-SA, not sure about its attribution requirements). Did you intentionally choose CC, despite of this? Or should you perhaps reconsider the license?

Spirik, 2018-01-08:
Hmmm... At this point I am not sure why I chose this license over the other options in the first place (I used it for my previous [library](https://github.com/Spirik/KeyDetector) as well). Although, by the looks of it, it should be sufficient in this particular case, at least for the time being. However in the light of [this discussion](https://opensource.stackexchange.com/questions/1717/why-is-cc-by-sa-discouraged-for-code), I may reconsider lately.

One thing certain, I should've done my initial research better.

Thank you for heads up on this!

Spirik, 2018-01-08:
Ok, let's give it a try. Migration to LGPL-v3: #3.

Spirik, 2018-01-08:
And [done](https://github.com/Spirik/GEM/blob/master/LICENSE.txt)!

closed 2018-01-08 by Spirik