Spirik/GEM #8: u8g2, Adafruit GFX libraries

AlexFundorin, 2019-06-15:
Your menu looks great.
Please, add support for these two major GFX libraries. Or, at least, for u8g2. Thanks in advance.

Spirik, 2019-06-16:
Hi @AlexFundorin !

Adding support for most popular LCD libraries would be great and seems like the only natural way of further developing GEM library. It is indeed in scope of my work, however not in the nearest future, I am afraid. But I will definitely will look into this as soon as I've got a chance!

Thank you for your interest in GEM!

AlexFundorin, 2019-06-16:
What did you use to demonstrate the LCD animation?

Spirik, 2019-06-16:
You mean in the docs? Just real-life photos glued together into gif:)

Spirik, 2020-05-07:
@AlexFundorin, hey there!

Just a little heads-up, that I've added support for U8g2 library (in the form of a separate class `GEM_u8g2`), and updated the documentation accordingly (readme, [wiki](https://github.com/Spirik/GEM/wiki ) and examples). Updates available since [release 1.1.0](https://github.com/Spirik/GEM/releases/tag/1.1.0). Feel free to check it out, if you are still interested=)

fmafma, 2020-05-08:
Hi!
I was looking for a good menu library, and didn't select GEM as it didn't support u8g2. But I just saw today that it now does! This is great :o)

But it seems that it still import the AltSerialGraphicLCD lib (from GEM.h file)... Would it be possible to avoid the install of that lib, when using u8g2?

Thanks,

Spirik, 2020-05-08:
@fmafma, sure!

What you want is just to use `#include <GEM_u8g2.h>` in your sketch, and remove `#include <GEM.h>`. Don't worry, AltSerialGraphicLCD version (although being shipped with library) won't increase the size of your compiled sketch - you'll be using only `GEM_u8g2` features (given that you'll include only U8g2 version and not AltSerialGraphicLCD one at the same time).

fmafma, 2020-05-08:
Mmm, I installed GEM from Arduino IDE, and I don't have AltSerialGraphicLCD installed. And when I install AltSerialGraphicLCD manually, I also have to install ESPSoftSerial, as I develop for ESP32...

What I had to do is delete GEM.h/cpp files, and remove the GEM.h include from a couple of other files.

fmafma, 2020-05-08:
Sorry, I had to replace GEM.h include with GEM_u8g2.h in these files...

Spirik, 2020-05-08:
@fmafma Oh, now I see where the problem is. Indeed there are some direct imports of GEM.h left in supplementary classes. I will rework that a little. Meanwhile you can use your workaround for a time being.

Spirik, 2020-05-08:
@fmafma, I've removed direct includes of GEM.h from the supplementary classes (so no need to replace them with includes of GEM_u8g2.h any more in your case).

However, both libraries are still required during compilation (due to how compiler works). So (as in your case) if you don't want to install AltSerialGraphicLCD, you would need to either remove GEM.h and GEM.cpp files or make them blank.

I am not happy with this solution, of course. Will try to find more elegant one in the future.

These changes available since [release 1.1.1](https://github.com/Spirik/GEM/releases/tag/1.1.1) through Library Manager (as soon as it will grab and publish them) or as a direct download from GitHub.

fmafma, 2020-05-08:
Thanks!
I started to play with GEM, and I like it very much. I will have some suggestions for improvements; I will come back later.

Spirik, 2020-05-08:
@fmafma Awesome!

Spirik, 2020-05-09:
Closing this to encourage creation of separate issues for future problems/suggestions.

closed 2020-05-09 by Spirik