Spirik/GEM #12: Use GEM as optionnal menu

fmafma, 2020-05-10:
Hi!

In my application, I plan to use the screen to display some informations (using u8g2). And, sometimes, I want to switch to GEM menu system to do some configuration and specific tasks, then go back to my own display.

So, I need to enable/disable the menu system on request.

How would you do that?

Thanks,

Spirik, 2020-05-10:
The way I would do that is by [creating](https://github.com/Spirik/GEM/wiki/How-to#create-button-advanced-with-custom-context) a [context](https://github.com/Spirik/GEM#appcontext) for your primary application, similarly to how it is done in [Blink](https://github.com/Spirik/GEM/wiki/Example-02%3A-Blink) and [Party Hard](https://github.com/Spirik/GEM/wiki/Example-03%3A-Party-Hard%21) examples. But instead of launching it manually through the menu button, invoke it right upon the sketch loads, e.g. in `setup()` right after `menu.init()` and _instead of_ `menu.draw()`. That way, instead of going straight to the menu upon loading, your primary task will be launched. And later you can draw your menu on screen by exiting the context (and re-entering it after changes would be made to the configuration through the menu, etc.).

Does that make sense? I probably can elaborate a little more and prepare some barebone example, based on, say, Blink or Party Hard examples.

fmafma, 2020-05-10:
I see! But a little working example would be nice :o)

Spirik, 2020-05-11:
@fmafma Here is [a gist](https://gist.github.com/Spirik/9d41973b13b6759ce0a2f8c5694f8771), based on Blink example.

I've denoted some important changes (and notes) with `!!!` in comments.

So basically, Blink routine will be launched as soon as device starts (by calling `blinkMillis()` from a `setup()` instead of `menu.draw()`). The rest is essentially the same as in original example: you detect keypresses in `blinkMillisContextLoop()`, and if `GEM_KEY_CANCEL` is pressed, draw the Configuration menu to the screen (you can implement your own key detection mechanic there, of course).

Hope that helps. Let me know if you have questions!

fmafma, 2020-05-11:
Thanks, I'll try that!

fmafma, 2020-05-11:
Ok, it works fine. I can totally go this way. Thank you.

closed 2020-05-11 by Spirik