Spirik/GEM #13: Refresh GEM when not using internel key scrutation

fmafma, 2020-05-11:
Hi!

I'm using an [I2cNavKey](https://github.com/Fattoresaimon/I2CNavKey) to navigate the menu. This system has an interrupt pin when a key changes, so I only call registerKeyPress() in this case.

But in order to GEM to refresh its loop(s), it seems that readForKey() as to be called very often. Am I right? So, can I call it even if I don't have any key binded to GEM nor u8g2? Or is there another method to call to refresh GEM internals?

Thanks,

Spirik, 2020-05-11:
You can use your [own solution](https://github.com/Spirik/GEM/wiki/How-to#control-menu-from-your-sketch) for key press detection, just make sure to pass correct key codes to `registerKeyPress()`. And yes, you can do this outside of the generic loop, so, basically, solution with interrupts (as in your case) should work fine. At least I am not seeing any problems with that at the moment.

Just run inside your interrupt handler:
```cpp
byte keyCode = GEM_KEY_DOWN; // Get correct key code here
if (menu.readyForKey()) {
  menu.registerKeyPress(keyCode);
}
```

fmafma, 2020-05-11:
Got it. Thanks.

closed 2020-05-11 by Spirik