Spirik/GEM #14: Readonly items

fmafma, 2020-05-11:
What about:

- skip readonly items when navigating the menu;
- extend this feature to all items, but GEM_ITEM_BACK.

Spirik, 2020-05-11:
Hm, thank you for suggestion!

However, it may potentially lead to some edge cases, which I'd rather not be messing with. E.g. from the top of my head, what if all of a second screen of the menu (in case menu items overflow and there is vertical scroll) is occupied by readonly items? Then user wouldn't be able to navigate to it, because he would always be back to the start of the first screen (while scrollbar would be still visible to him, confusingly enough). Solution would require additional logic and program space, which is a of a concern, especially for smaller AVR controllers.

That being said, I see your point though! So let's add it to back burner for now, and I will think about it a little more.

fmafma, 2020-05-11:
About skipping readonly entries, I agree, this may be confusing in the case you describe..

But I think buttons and links should respect the readonly flag, like values, as they do have the setReadonly() method.

Spirik, 2020-05-11:
That's a good point. `setReadony()` essentially has no effect on buttons and links in current version. Probably I can mark them with `^` the same way other readonly menu items are marked and prevent action from triggering when selecting them.

fmafma, 2020-05-11:
Looking at the code, this should not be very complicated ;o)

Spirik, 2020-05-11:
True=)

fmafma, 2020-05-11:
I made a quick and dirty change: I switch to GEM_POINTER_DASH for readonly items, and stay on GEM_POINTER_ROW for editable items; not bad.
A better solution should be to have 2 versions of the cursor for each DASH/ROW. For example, for ROW, we could use a rectangle around readonly items, instead of having the entire line video reversed.
Don't know what alternative to use for DASH, though...

Spirik, 2020-05-11:
Yeah, I thought about that too. Probably dotted appearance for DASH version will do the trick.

fmafma, 2020-05-11:
Or just change the width (enlarge to 3 for editable, and only 1 for readonly)?

Spirik, 2020-05-11:
That may be too similar to the normal state of the pointer. But at least it is easier to try)

Spirik, 2020-05-11:
Readonly modes for menu links and buttons will be available since [release 1.1.2](https://github.com/Spirik/GEM/releases/tag/1.1.2).

Spirik, 2020-05-11:
@fmafma Created draft for readonly versions of menu pointers. You can check it out [here](https://github.com/Spirik/GEM/blob/24f9c611d0a05306dcadfc7312cbba770fec4929/src/GEM_u8g2.cpp#L386) (it is not in master yet). It is checker pattern for `GEM_POINTER_DASH` (happy with that) and rectangle for `GEM_POINTER_ROW` (not so much for some reason).

fmafma, 2020-05-11:
Checker pattern is great!

For the rectangle, what about:

`_u8g2.drawHLine(0, pointerPosition-1, _u8g2.getDisplayWidth()-1);`
`_u8g2.drawHLine(0, pointerPosition+m_menuItemHeight, _u8g2.getDisplayWidth()-1);`
`_u8g2.drawVLine(0, pointerPosition-1, _menuItemHeight+2);`
`_u8g2.drawVLine(_u8g2.getDisplayWidth()-2, pointerPosition-1, _menuItemHeight+2);`

Spirik, 2020-05-11:
I think I've found pleasing enough solution (to my taste, that is) for ROW pointer in combining checker pattern with default filled row. (Not yet tried your latest suggestion though). Pushed it to the same branch.

[image]

Spirik, 2020-05-12:
Checker pattern versions will be available since release [1.1.3](https://github.com/Spirik/GEM/releases/tag/1.1.3).

closed 2020-05-13 by Spirik