Spirik/GEM #18: New val types

fmafma, 2020-05-12:
For now, only GEM_VAL_INTEGER and GEM_VAL_BYTE are supported. What about adding GEM_VAL_FLOAT?

It would also be very handy to have a GEM_VAL_HEX type, showing int/byte values in hexa, but it may be an option for those 2 types; don't know what is better...

I think the types management could be refactored, to support new types, with there specific edition way. For example, I'm thinking to a GEM_VAL_IP, to edit an IP value. But I guess it is a lot of work.

Spirik, 2020-05-12:
I intentionally tried to keep library size relatively small, implementing only subset of tools, that can be utilized by end user to implement their own rules for data inputs (to some extent). Hence validation callbacks. One can implement IP address input (and HEX one as well) based on `char[]` array and subsequent validation of a resulting string on their end. That way other users won't have to deal with bloated library with features they most likely won't use.

As to `float` values - it would be nice to have those, I completely agree. But I ultimately decided not to include their support in the initial releases of GEM due to (you guessed it) library size: using `atof()` quite significantly increases sketch size (for at least 7% of storage space on Arduino Uno). But I will eventually implement it anyway, I guess. For now, one can use different units to represent their values (e.g. ms instead of seconds) and, if absolutely necessary, use callbacks to convert their string input into floats through `atof()` (as).

fmafma, 2020-05-12:
I understand. The ultimate way would be to have #define blocs, to only include needed parts.
As I now develop on ESPs, flash/memory is not an issue, but small AVR are quickly saturated!

closed 2020-05-12 by Spirik

Spirik, 2020-11-21:
Good news: optional support for floating-point variables (`float` and `double`) added since version 1.2.0 (and is enabled by default), see [new section](https://github.com/Spirik/GEM#floating-point-variables) in Readme for implementation specifics.