Spirik/GEM #20: Redraw the menu after a button action

pailpoe, 2020-11-18:
Hello,
I have a button with a text : “push to save”
It’s possible when the button is pressed to change the text with “save ok” and update the screen. 
Thanks a lot for your work ! , my project use your library ( I add the float support ).

pailpoe, 2020-11-18:
My project : [https://github.com/pailpoe/MiniThread]

Spirik, 2020-11-18:
@pailpoe Hello there!

I actually saw your amazing work (and went as far as digging through your [forum post](https://www.usinages.com/threads/minithread-dro-vis-mere-electronique-pour-tour.117735/page-7) :) ), and float implementation in particular=) It inspired me to do my own implementation (it is very similar to yours). It will be available soon as an update, right after I finish updating docs. Currently available in a [float-support](https://github.com/Spirik/GEM/tree/float-support) branch. I had to workaround some differences between SAMD and AVR implementation, and added some control over whether to include `float` and `double` support in the build or not (via `config.h` file) to save some space in cases where such support is not required. But that's for another topic=)

As for your question: currently title of the `MenuItem` is private and there is no way to update it from external sketch. However, I think I can add some public method for `MenuItem` to allow user to update title dynamically. That doesn't seem like an essential feature by any means, but I can see some cases where it can be useful (e.g. your particular case).

Then you will be able to update title of the menu item in your callback function (like so, presumably, `myButton.setTitle("save ok")`), and call `menu.drawMenu()`. However, that will be your job to implement some delay to change its title back afterwards. Then, after required delay has passed, you can set title back (`myButton.setTitle("push to save")`) and upon exiting your callback menu will be redrawn automatically.

Does that sound like what you want to accomplish, or have I misunderstood something?

I can look into it right after I finish publishing `float` implementation.

pailpoe, 2020-11-18:
Hello,
Good news if you officially support float ! 
Yes, i want to dynamically change the title  of a item (button) . I want to change the title of the button save when a settings change in the menu with "Need to be save" and after the saving "Save ok". 
Your work is amazing ! and thanks a lot for your work !

Spirik, 2020-11-21:
@pailpoe, hi!

Just published version of GEM (v1.2.1) with native support for floating-point variables (`float` and `double`), see [new section](https://github.com/Spirik/GEM#floating-point-variables) in Readme for implementation specifics.

And more relevant to this discussion: I've added two new methods `GEMItem::setTitle()` and `GEMItem::getTitle()` to dynamically change title of menu items. And here's [demonstration](https://www.youtube.com/watch?v=njeiM2M-e_A) of Save button example. Code for this example can be found [here](https://gist.github.com/Spirik/42793033f5009c2826a3ca9ac62120b2).

Does that look similar to what you want to implement?

pailpoe, 2020-11-22:
Thanks a lot,
I like your implementation with the possibility to set the precision for the float and the button with the possibility to change the title is perfect ! 
I test this new lib as soon as possibile. 
I open a new enhancement for a visible or not option for a item.

closed 2020-11-24 by Spirik