Spirik/GEM #23: Visible or NotVisible function

pailpoe, 2020-11-22:
Hello Alexander,
In my project, I use a check box "Use motor" with a lot of parameter for my motor for exemple "Option 1",.... 
When "Use motor" is don't check, i use the read only function to lock the parameter but i will be more clean to don't see these parameter with a visible function. 
example : 
MyMenuItem.visible(false);
....

Spirik, 2020-11-22:
Yeah, I thought about it, as well as an ability to dynamically add and destroy menu items. The thing is, current algorithm used to draw and navigate through menu items relies heavily on the fixed (and known beforehand) number of menu items contained in a menu page. To implement aforementioned features I would need to rewrite this simple algorithm, probably significantly (probably not - not sure yet). So I am not determined to do that at the moment. Probably I will look at it again some time later. I completely agree, that this is useful feature to have.

For now, using `setReadonly` method is advised. You can also use additional menu pages to group specific menu items which are relevant only when specific option is set (so you can set read only mode for the whole menu page instead of each menu item individually).

My current plan for the next update is to implement option to choose which version of GEM (`AltSerialGraphicLCD` or `U8g2` based) should be compiled, based on the inclusions in `config.h` file (which is currently used to control optional support for floating-point variables). That way, there won't be requirement to have both of the supported graphics libraries installed in the system at the same time (regardless of which one is actually used), and you won't have to delete `.h` and `.cpp` files of not needed version from the source code manually.

pailpoe, 2020-11-22:
Ok, I understand the difficulty to add a visible option. For the config file, it's a good idea, I need to delete the link to AltSerialGraphicLCD.

Spirik, 2020-11-24:
Just added an ability to choose which version of GEM library (`AltSerialGraphicLCD` or `U8g2` based) should be compiled via [config.h](https://github.com/Spirik/GEM/blob/master/src/config.h) file. Available since ver. 1.2.2.

Now I'll dig into option to hide menu items=)

pailpoe, 2020-11-24:
Good news ! 
I try this release this weekend.

Spirik, 2020-12-05:
Added ability to hide/show menu items in [ver. 1.2.4](https://github.com/Spirik/GEM/releases/tag/1.2.4)!

To hide menu item you can call:
```cpp
myMenuItem.hide();
myMenuItem.hide(true);
```

To show menu item you can call:
```cpp
myMenuItem.hide(false);
myMenuItem.show();
```

To get current visibility status:
```cpp
boolean isHidden = myMenuItem.getHidden();
```
Hidden menu items won't be printed to the screen the next time menu is drawn. Note that if you call these methods from within menu item callback, menu will be redrawn automatically (so no need to call `myMenu.drawMenu()` manually after menu item visibility is changed).

pailpoe, 2020-12-06:
It's perfect, thanks !
[Video](https://www.youtube.com/watch?v=lOuFHhvQdBU)

closed 2020-12-06 by Spirik