Spirik/GEM #26: Gem 1.2.2 (itoa and settitle for pageitem)

pailpoe, 2020-11-25:
Hello, 
I put your new release (1.2.2) tonight and itoa is not define with my configuration (stm32 core). 
i need to add this 
`#include <Arduino.h>
#include "GEM_u8g2.h"
#include <itoa.h>

#ifdef GEM_ENABLE_U8G2_VERSION`
This release is perfect, with the possibility to change the resolution for float. 
I use the setTitle for change the lang of the menu but it's impossble tu change the name of a pageitem ! 
a small video : 
[https://www.youtube.com/watch?v=-S_ZRFjv2ss](url)

Great job and thankyou

Spirik, 2020-11-25:
You're project looks amazing, thank you for sharing the video!

Yes, I indeed didn't add an ability to change title of the menu page - didn't really thought, that one can use this dynamic titles for localization purposes (which is an awesome idea, btw)! That was an oversight on my part. Will add this ability in next release (will try to make it quickly).

As to exlplicit  inclusion of `itoa.h` - that must be an STM32 requirement (much like an inclusion of `avr/dtostrf.h` is required for non-AVR boards). Never got a chance to test it on actual STM board (only have an ATmega328 and M0 boards), so thank you for finding that out! Will add in the next release as well.

Once again, thank you for sharing your project! Keep up the great work 👍

Snake is dope 😎

Spirik, 2020-11-26:
@pailpoe Just rolled out fixes in [1.2.3 release](https://github.com/Spirik/GEM/releases/tag/1.2.3) (both dynamic menu page titles and better `itoa.h` support). Please, let me know how this will work for you.

pailpoe, 2020-11-27:
I just add this release and it's perfect ! Great job.

closed 2020-11-28 by Spirik