Spirik/GEM #29: item title for a button with a submenu too short

pailpoe, 2020-11-28:
Hello,

I see that the title displayed for a button with a sub menu is 17 char max before trimming and can be 18 normally. 
Just a remark . 
I use the normal size for the menu. 

Best regards

Spirik, 2020-11-28:
Do you use [default](https://github.com/Spirik/GEM/wiki/Menu-appearance) menu size settings? Just checked, it's 19 characters for me (for submenu link and button titles). Or maybe I didn't understand you correctly.

[image]

pailpoe, 2020-11-28:
Hello,
I see this 
[image]
My configuration : menu(u8g2,GEM_POINTER_ROW, 5,10,10,75)
It’s not a problem but just a remark

Spirik, 2020-11-28:
Oh, I see where the problem lies.

I don't store full length of the available space for full title. Instead, it is calculated as a sum of two lengths: length of the menu item title and length of the value. Which, in turn, are calculated as a function of horizontal dimension of the screen and left offset of the value (which equals 86 by default and 75 in your case). The thing is, that results of this calculations are stored as a `byte` (mainly to save space), so in case of floating-point result of calculation, it will be simple truncated (not even rounded), leaving only the integer part of the number. So that may result in full title lengths being 1 character short in some edge cases. (I'd prefer this than overflowing actually, so I'm personally fine with this).

The good news is that in most cases it is easily fixable=) So, in your particular case, just set 74 instead of 75. That should do the trick;)

pailpoe, 2020-11-28:
Thank you, I will try

closed 2020-11-30 by Spirik