Spirik/GEM #31: From the main menu, if we press cancel, Launch a action from a item

pailpoe, 2020-12-06:
Hello Alexander,
It's possible to add the possibility from the main menu when i press cancel to launch a item ( i want to launch my main context without the need to scroll up and click ok).

Thanks.

Spirik, 2020-12-06:
That's an interesting idea! I see, how it can be useful. I will look into it.

Spirik, 2020-12-12:
Added ability to specify an optional `exitAction` callback in `GEMPage` constructor in [ver. 1.2.5](https://github.com/Spirik/GEM/releases/tag/1.2.5)!

```cpp
GEMPage menuPageMain("Configuration", blinkMillis); // Main page with exit callback, that will be fired when pressing Cancel button
```

This can be any action, including those in which context is created. Action will be fired only for top-level menu page (i.e. menu page that has no parent) when not in edit mode and `GEM_KEY_CANCEL` is pressed.

Gist with auto-loaded context updated accordingly and can be found [here](https://gist.github.com/Spirik/9d41973b13b6759ce0a2f8c5694f8771).

There is slight difference between launching an action via button and launching an action through an exit callback: when launching an action via button current menu pointer position remains unchanged (i.e. button will still be in focus when menu is redrawn to the screen next time), whereas in case of an exit callback menu pointer position will be reset (i.e. set to the first menu item).

pailpoe, 2020-12-16:
Thankyou, i will test this modification this week.

pailpoe, 2020-12-25:
It’s perfect ! 
Happy Christmas

Spirik, 2020-12-25:
Glad to hear!

Merry Christmas=) 🎅🎄❄

closed 2020-12-25 by Spirik