Spirik/GEM #33: Add switch to disable splash screen and/or change itd display time

JakubRakus, 2021-02-27:
I think there should be a possibility to disable splash screen and/or to change display time of it. Something like this (pseudocode):
```
void GEM::init(bool show_splash, uint32_t delay_ms)
{
    init_stuff_here();
    if(show_splash)
    {
        splash_XBMP_display();
        if(_enableVersion)
        {
            delay(delay_ms/2);
            version_display();
            delay(delay_ms/2);
        }
        else
        {
            delay(delay_ms);
        }
    }
    clear_stuff_here();
}
```
BTW: using hard-coded `delay()` or any `do_nothing_for_some_time` inside library functions are strongly discouraged in embedded software unless they are explicitly named (e.g. `init_splash_wait`), it could break the whole app, especially those not RTOS-based!

Spirik, 2021-02-28:
That's a good suggestion. For now you can "disable" splash by drawing empty sprite on the screen, but still it will lead to 1000ms delay, so it's probably better to add explicit option to disable splash whatsoever, as you suggested. Will look into it.

I assume, adding option to specify delay time (e.g. `GEM::setSplashDelay()`) will be enough, so that setting it to 0 (and adding explicit check for that in the code) will lead to skipping splash drawing routine.

Spirik, 2021-02-28:
Here is PR #34 with `GEM::setSplashDelay()` and `GEM_u8g2::setSplashDelay()` methods to explicitly set duration of time splash screen will be visible. Setting to 0 will disable splash screen.

JakubRakus, 2021-02-28:
Looks very good to me, I've reviewed and accepted changes in Your PR.

closed 2021-02-28 by JakubRakus

Spirik, 2021-02-28:
Thank you!

I've merged PR and created [new release](https://github.com/Spirik/GEM/releases/tag/1.2.6). Should be available via Arduino Library Manager within an hour or two.