Spirik/GEM #36: Use with Rotary Encoder?

polygonfuture, 2021-09-04:
Hi there,

Trying to find examples of using GEM using a single push-button rotary encoder.  Do you have any examples of this?

Cheers

Spirik, 2021-09-04:
Hi=)

Unfortunately there is no example for rotary encoders specifically, but theoretically as soon as input device can produce 6 basic operations that menu understands (up, right, down, left, ok, cancel) it should be completely possible. But it is up to you to figure out how to extract those readings from encoder (or any other input device for that matter).

See [Control menu from your sketch](https://github.com/Spirik/GEM/wiki/How-to#control-menu-from-your-sketch) section of wiki for a little more details. You may want to look into how [`KeyDetector`](https://github.com/Spirik/GEM/wiki/How-to#detect-key-presses-with-keydetector) or [`U8g2`](https://github.com/Spirik/GEM/wiki/How-to#detect-key-presses-with-u8g2) key detection organized in supplied examples (in their `loop()`) as a starting point.

Hopefully this may help!

polygonfuture, 2021-09-04:
Ok understood.   It looks like I may just want to bypass u8g2's key detection, since the method to detect an encoder pulse is not a high low state, but a specific value increasing or decreasing.   

So in the case of a single encoder, we really only have a left, right, or button press as inputs.   So in this case are there any negative consequences form using a single input (rotary left) to both GEM's LEFT and UP values?   Same goes for rotary right?

Spirik, 2021-09-05:
How an item would be saved in that scheme (apart from pressing cancel button, which results in exit without saving), since you're planning to use push button to select digit position while in edit mode? Or is your plan to save an item any time its digit is saved? That may lead to some unwanted results (unnecessary triggering validation callbacks, redrawing menu, etc.).

And I am afraid there is no way you can edit this logic without editing source code of library.

What you can do without modifying the source code, however, is to wait for menu to be ready to accept a key press and then fire up your own logic to determine which menu key press to trigger:

```cpp
void loop() {
  // If menu is ready to accept button press...
  if (menu.readyForKey()) {
    // ...detect key press using your own logic
    byte keyCode = myRotaryEncoderKeyDetectionLogic();
    // Pass pressed button to menu
    menu.registerKeyPress(keyCode);
  }
}
```

polygonfuture, 2021-09-05:
Well the encoder has a push button built in.  So I would have two push buttons total, and two left/right keys (encoder).

This sounds promising.  I'll take a stab at this in the morning.

Another question:  How does one constrain the values while editing?  
For instance lets say I have a value of 3523 and I want to limit its settings from 3000 - 3800.   Is there a way to handle constraining this value?

Also related, is there possibly a way to instead of editing digit by digit, you could simply +/- from the full value all together?

Spirik, 2021-09-05:
> Another question: How does one constrain the values while editing?
> For instance lets say I have a value of 3523 and I want to limit its settings from 3000 - 3800. Is there a way to handle constraining this value?

You can not constrain user input during edit mode, however you can implement validation [callback](https://github.com/Spirik/GEM/wiki/How-to#add-menu-item-callback), that will be triggered upon saving the value: that way you can change value to, say, the closest one from the valid range, once user finished editing.

> Also related, is there possibly a way to instead of editing digit by digit, you could simply +/- from the full value all together?

Nope, unfortunately there is no way to do that at the moment. However there is possibility it will be implemented in the future via dedicated menu item type.

zandernc, 2023-07-10:
I implemented this as you suggested, Spirik, assigning the encoder to up/down keypresses, and a short press of the button to "ok" and a long press to "right". Works great. Thank you!

Spirik, 2023-07-11:
@zandernc Is it possible to rotate the knob while simultaneously being pressed on your encoder? If it is, then I would suggest to consider the following scheme:
* Rotation - Up, Down;
* Rotation + Press - Left, Right;
* Short press - Ok;
* Long press - Cancel.

That way we will cover all 6 interactions GEM is able to respond to.

Spirik, 2023-09-28:
👋  FYI: just updated GEM with [examples](https://github.com/Spirik/GEM/wiki/Example-05%3A-Encoder-U8g2-version) on how to control menu with rotary encoder using latest version of KeyDetector (implementing control scheme described above).

closed 2023-11-20 by Spirik