Spirik/GEM #43: menuItemNext being a wild pointer in certain circumstances

shezik, 2021-11-06:
I'm working on an ESP32 project where everything GEM-related is dynamically allocated.
However LoadProhibited fatal error occurs when *menu.drawMenu()* is called. Turns out that pointer *menuItemNext* defined in *GEMItem.h* is wild and the issue disappears if *menuItemNext* is initialized with *nullptr*.

```
    GEMPage* parentPage = nullptr;  
    GEMPage* linkedPage;  
    GEMItem* menuItemNext = nullptr;  // <- Here
    void (*buttonAction)();
```

I wonder if this is because that all objects are dynamically allocated or just that I'm retarded.
Code can be provided if required. :)

Spirik, 2021-11-06:
Good catch!

Yes, by all means `menuItemNext` should be initialized with `nullptr` by default. However, this error doesn't trigger during "normal" workflow (e.g. examples that come with GEM) and probably related to dynamic allocation in your case. And indeed the last menu item of the menu page may end up having `menuItemNext` left as an uninitialized wild pointer. Which is, of course, bad and should be avoided. So I will fix it on the next patch release. Thank you!

Spirik, 2021-11-07:
Fix now available in [1.3.4](https://github.com/Spirik/GEM/releases/tag/1.3.4) release. Please check, if this will work for you.

shezik, 2021-11-07:
> Fix now available in [1.3.4](https://github.com/Spirik/GEM/releases/tag/1.3.4) release. Please check, if this will work for you.

It definitely does. Thank you for being so responsive! :D

closed 2021-11-07 by shezik