Spirik/GEM #49: .drawMenu() leads to blinking screen

gerberli, 2022-02-09:
Hi

Thanks for this absolute blast of a library! I really appreciate the extensive documentation.

I'm trying to make a menu with GEM_adafruit_gfx.h on [this ](https://www.adafruit.com/product/1431) screen and want to display a value that automatically updates (e.g. the runtime millis()) as a MenuItem. My problem is now that the value does not update unless I press one of the buttons. I tried to call menu.drawMenu every certain time interval, but this leads to extensive blinking of the screen (i.e. going to black and then correctly displaying the Menu). 

Obviously there must be a way to update the screen without blinking, as this happens when a key is pressed. I tried to understand what exactly happens when pressing a button, but could find nothing else but .drawMenu(). What am I missing?
How can I update the screen (or displayed values), without the blinking?

Cheers,
gerberli

Spirik, 2022-02-09:
Hi, gerberli!

Glad you're liking the library=)

Unfortunately, the only way to refresh screen from within user sketch is to call `drawMenu()` method. Internally, to increase performance, sometimes only part of the screen is drawn/redrawn (e.g. during edition of a variable), hence there is no visual blinking. But that relies on the calculations of current position of menu pointer, so is not practical for most end user implementations (and for your case in particular).

However, I may suggest looking into some (rather hacky) solutions to achieve this: after menu is drawn to the screen, it is possible to draw something else on top of it (from the user sketch using Adafruit GFX library), so you can draw and redraw your millis variable on top of its place on menu page. But, apart from exact calculations of menu item position, that will require monitoring the current state of the menu: to react to menu page changes and possible collisions with menu pointer (for that you, probably, may want to do this menu item [readonly](https://github.com/Spirik/GEM/wiki/How-to#make-menu-item-readonly) and change menu pointer appearance to `GEM_POINTER_DASH`).

Or, possibly, you can draw your millis variable outside of the menu altogether, by adjusting `menuItemsPerScreen_` or `menuPageScreenTopOffset_` parameters of the `GEM_adafruit_gfx` constructor, to let some free space under menu, or above it (while making sure your variable won't clash with menu page title).

I intend to add more convenient way to monitor current state of the menu object some time in the future (menu pointer position, currently active page, etc.), so it will make implementation fo some this "hacky" modifications a little bit easier. But for now that is all, that comes to mind.

gerberli, 2022-02-09:
Hi Spirik

Thanks a lot for the quick reply. 
I understand. I was thinking about your suggestion as well, having a dedicated space where GEM does not write and just manually write to that with Adafruit_GFX.

Thanks for the help!

closed 2022-02-14 by Spirik