Spirik/GEM #51: Problem with floats

baettigp, 2022-06-05:
Hi, thank you very much for your library. I am trying to implement it in a small project.

-I have problems setting a float in the menu. 
I would like to set the boiler temperature between 85 and 96 degrees C in 0.1 degree steps
So I call it as follows:

```
float BoilerTemp =92.0;

void validateBoilerTemp();
GEMItem menuItemBoilerTemp("Temp degC:", BoilerTemp, validateBoilerTemp);

  menuPageSettings.addMenuItem(menuItemBoilerTemp);

void validateBoilerTemp() {
  
  // Check if mtime variable is within allowable range (i.e. 10..60)
  if (BoilerTemp < 85.0) {
    BoilerTemp = 85.0;
  }
  if (BoilerTemp > 96.0) {
    BoilerTemp = 96.0;
  }
```

When I go into the settings menu, the line reads "Temp degC 882.7" or after I enter the second time "Temp degC -242.0", far outside of the validated range from 85.0 to 96.0.

If I change it, sometimes it sets to 85, but as soon as I leave and reenter the menu, it's back at 882.7.

-Btw, How can I call the method setPrecision for floats? At the moment I manually set it to 1 digit in config.h, but I don't need all the floatsI want to represent only with 1 digits

Thank you very much for your help, if you need any more info, code, let me know.

Spirik, 2022-06-06:
Hello! Thank you for using GEM!

What platform do you use? Just tried the following on Arduino UNO (with Adafruit GFX library) and it works:

```cpp
float numberFloat = 92.0;
void validateFloat();
GEMItem menuItemFloat("Float:", numberFloat, validateFloat);

...

menuPageMain.addMenuItem(menuItemFloat);
menuItemFloat.setPrecision(1); // Set precision

...

void validateFloat() {
  // Check if numberFloat variable is within allowable range (i.e. >= 85.0 and <= 96.0)
  if (numberFloat < 85.0) {
    numberFloat = 85.0;
  }
  if (numberFloat > 96.0) {
    numberFloat = 96.0;
  }
  // Print numberFloat variable to Serial
  Serial.print("numberFloat set: ");
  Serial.println(numberFloat);
}
```

Also note call to `menuItemFloat.setPrecision(1);` for setting precision.

baettigp, 2022-06-06:
Hi AlexanderThank you very much! I am using an Arduino due. Might the implementation of sprtof be different? I remember reading something a long time ago.Best regards, Pio--Sent from my Android phone with GMX Mail. Please excuse my brevity.

Spirik, 2022-06-06:
Arduino Due uses Cortex-M3 CPU and so indeed doesn't support `dtostrf` [by default](https://github.com/Spirik/GEM#floating-point-variables). However, in GEM it is included explicitly for SAM (like Arduino Due) and SAMD (like Arduino Zero) boards, see [here](https://github.com/Spirik/GEM/blob/master/src/GEM_adafruit_gfx.cpp#L43) for example. And since you don't get any error warnings regarding `dtostrf` not being declared, presumably it is included successfully.

Unfortunately I don't have Arduino Due at hand to test it myself, however I've tested this example on Arduino Zero (Cortex-M0+ based SAMD board) and it works.

I may suggest writing simple sketch with only one float variable (for testing purposes) and probably try compiling it for different board (if you have any).

closed 2024-05-10 by Spirik