Spirik/GEM #56: UTF8 with u8g2

TimonPeng, 2022-08-26:
Hi Alex!

It's me again! The design and structure of GEM is truly excellent! I love it!

When I try to enable UTF8 print like the u8g2 example, but seems can't works.

https://github.com/olikraus/U8g2_Arduino/blob/dd729c5253eb85d907f0f84a1c53074bffb7a43a/examples/page_buffer/PrintUTF8/PrintUTF8.ino#L338

My codes:
```
GEMItem menuItemInt("数字:", number);

void setup() {
  Serial.begin(115200);

  u8g2.begin();
  u8g2.enableUTF8Print();
  // https://github.com/larryli/u8g2_wqy
  u8g2.setFont(u8g2_font_wqy12_t_gb2312);

  menu.init();
  setupMenu();
  menu.drawMenu();
}
```

TimonPeng, 2022-08-26:
It's because GEM have default internal `_fontFamilies`, so do you have any plan to support external fonts setting when init? for main title, items and content. Thanks.

Spirik, 2022-08-26:
Oh yeah, I am definitely considering it for the future updates.

For the time being the best way to workaround this is to [enable Cyrillic](https://github.com/Spirik/GEM/wiki/How-to#enable-cyrillic) support via `GEM_u8g2::enableCyrillic()` [method](https://github.com/Spirik/GEM#methods) - that will [internally trigger](https://github.com/Spirik/GEM/blob/master/src/GEM_u8g2.cpp#L155) `.enableUTF8Print()` method of U8g2, thus activating support for UTF8 fonts. Then you will need to manually set fonts you want instead of predefined Cyrillic ones through editing these two defines in a source code of GEM_u8g2.h: one for [big font](https://github.com/Spirik/GEM/blob/master/src/GEM_u8g2.h#L50) and second for the [smaller font](https://github.com/Spirik/GEM/blob/master/src/GEM_u8g2.h#L51). That should technically work=)

Thank you for the kind words!

TimonPeng, 2022-08-26:
I really love this project ~~and you~~! As I said to my friend, this project is full of engineering aesthetics and design aesthetics.

Would you be willing to set up a github sponsor, or something else like paypal or cryptocurrency stuffs? I'd like to give a little support to you and your masterpiece!

Spirik, 2022-08-26:
I have not really considered it, to be honest, but thanks for the tip, probably I will someday=)

Spirik, 2024-01-04:
Implemented better custom fonts support in release [1.5.1](https://github.com/Spirik/GEM/releases/tag/1.5.1).

* New methods `::setFontBig()` and `::setFontSmall()` of U8g2 and Adafruit GFX versions of GEM to set "big" and "small" fonts;
* Support for UTF8 fonts for U8g2 version of GEM via `GEM_u8g2::enableUTF8()` method (only for menu titles and menu item labels - editable strings with UTF8 characters are not supported).

See corresponding section on [wiki](https://github.com/Spirik/GEM/wiki/Menu-appearance#changing-fonts) for more details.

closed 2024-01-11 by Spirik