Spirik/GEM #57: Menu controlling via serial port

TimonPeng, 2022-08-26:
When there aren't enough components around to build a prototype, maybe this is a good way to test.

TimonPeng, 2022-08-26:
I have written a version that works for `u8g2`, almost completely copied from your [KeyDetector](https://github.com/Spirik/KeyDetector), just for everyone to use you can close this issue anytime.

`SerialKey.h`
```c++
#include <Arduino.h>
#include <U8g2lib.h>

#define KEY_NONE 0

class SerialKey {
 public:
  SerialKey(int up_ascii_, int right_ascii_, int down_ascii_, int left_ascii_,
            int cancel_ascii_, int ok_ascii_);
  byte trigger = KEY_NONE;
  byte current = KEY_NONE;
  void detect();

 private:
  int _up_ascii;
  int _right_ascii;
  int _down_ascii;
  int _left_ascii;
  int _cancel_ascii;
  int _ok_ascii;
};
```

`SerialKey.cpp`
```c++
#include "SerialKey.h"

#include <Arduino.h>
#include <U8g2lib.h>

SerialKey::SerialKey(int up_ascii_, int right_ascii_, int down_ascii_,
                     int left_ascii_, int cancel_ascii_, int ok_ascii_)
    : _up_ascii(up_ascii_),
      _right_ascii(right_ascii_),
      _down_ascii(down_ascii_),
      _left_ascii(left_ascii_),
      _cancel_ascii(cancel_ascii_),
      _ok_ascii(ok_ascii_) {}

void SerialKey::detect() {
  boolean pressed = false;

  if (Serial.available() > 0) {
    int incomingByte = Serial.read();

    // Serial.print("Received: ");
    // Serial.println(incomingByte, DEC);

    if (incomingByte == _up_ascii) {
      current = U8X8_MSG_GPIO_MENU_UP;
      pressed = true;
      // Serial.println("Pressed up");
    } else if (incomingByte == _right_ascii) {
      current = U8X8_MSG_GPIO_MENU_NEXT;
      pressed = true;
      // Serial.println("Pressed right");
    } else if (incomingByte == _down_ascii) {
      current = U8X8_MSG_GPIO_MENU_DOWN;
      pressed = true;
      // Serial.println("Pressed down");
    } else if (incomingByte == _left_ascii) {
      current = U8X8_MSG_GPIO_MENU_PREV;
      pressed = true;
      // Serial.println("Pressed left");
    } else if (incomingByte == _cancel_ascii) {
      current = U8X8_MSG_GPIO_MENU_HOME;
      pressed = true;
      // Serial.println("Pressed cancel");
    } else if (incomingByte == _ok_ascii) {
      current = U8X8_MSG_GPIO_MENU_SELECT;
      pressed = true;
      // Serial.println("Pressed ok");
    }
  } else {
    pressed = false;
  }

  if (!pressed) {
    current = KEY_NONE;
  }

  trigger = current;
}
```

Usage:
```c++
#include "SerialKey.h"

// w (119) -> ↑
// d (100) -> →
// s (115) -> ↓
// a (97) -> ←
// q (113) -> cancel
// e (101) -> ok
SerialKey serialKey(/*Up=*/119, /*Right/Next=*/100,
                    /*Down=*/115, /*Left/Prev=*/97,
                    /*Home/Cancel=*/113, /*Select/OK=*/101);

void loop() {
  if (menu.readyForKey()) {
    serialKey.detect();
    menu.registerKeyPress(serialKey.trigger);
  }

  delay(100);
}
```

Spirik, 2022-08-26:
Oh, that's great! I've just wanted to reply to an original post that it is totally possible to control GEM via Serial from the user sketch (and link to this [section of Wiki](https://github.com/Spirik/GEM/wiki/How-to#control-menu-from-your-sketch) for details) and you beat me to it with a complete example=)

Thanks for sharing, it can be useful for someone else! I will add link to this solution in wiki=)

Spirik, 2022-08-26:
If you change U8g2 specific aliases to more general GEM ones (e.g. `U8X8_MSG_GPIO_MENU_UP` -> `GEM_KEY_UP`, etc.) it will be possible to us this example with any versions of GEM (including AltSerialGraphicLCD and Adafruit GFX).

TimonPeng, 2022-08-26:
Okay no problem, do you mind if I just edit the file above or post the new one below here?

Spirik, 2022-08-26:
Feel free to edit your original post, that should be fine!

Spirik, 2022-08-26:
Seems to be fine, thank you!

Spirik, 2022-08-30:
Link to this example added to Wiki.