Spirik/GEM #58: Supports setting single one menu attributes

TimonPeng, 2022-08-27:
When I want to setting only one or a a little attributes, it need to fill complete instance here. Do you want to support a set function like `setMenuItemHeight` etc.?

before:
```c++
GEM_u8g2 menu(u8g2, /*menuPointerType_=*/GEM_POINTER_ROW,
              /*menuItemsPerScreen_=*/GEM_ITEMS_COUNT_AUTO,
              /*menuItemHeight_=*/12,
              /*menuPageScreenTopOffset_=*/10, /*menuValuesLeftOffset_=*/86);
```

after:
```c++
GEM_u8g2 menu(u8g2);
menu.setMenuItemHeight(12);
```

Spirik, 2022-08-27:
I can see that being convenient way of adjusting menu appearance on per parameter basis, but since the memory footprint is of a high concern (Example-03 barely fits on Arduino UNO in some versions of GEM) and this is generally done only once in the sketch, I figured that managing appearance simply through constructor arguments is a better way to go about it. So adding this is not of a high priority as of now, unfortunately. But at least I can run some tests to check how adding these new methods impacts compile size.

Spirik, 2023-12-28:
Digging up this thread=)

Recently ([1.5.0](https://github.com/Spirik/GEM/releases/tag/1.5.0)), added [`GEMAppearance`](https://github.com/Spirik/GEM#gemappearance) struct and `::setAppearance()` methods which allow to set appearance of menu pages individually (and even change at runtime).

Among other things, this makes it possible to change appearance parameters individually by editing corresponding properties of `GEMAppearance` object, like so:

```cpp
// Create menu object (its appearance settings will be populated with default values)
GEM menu(glcd);

...

void setupMenu() {
  ...
  // Create GEMAppearance object with general values (that will be used for every menu page if not overridden)
  GEMAppearance appearanceGeneral;
  appearanceGeneral.menuPointerType = GEM_POINTER_DASH;
  appearanceGeneral.menuItemsPerScreen = GEM_ITEMS_COUNT_AUTO;
  appearanceGeneral.menuItemHeight = 10;
  appearanceGeneral.menuPageScreenTopOffset = 10;
  appearanceGeneral.menuValuesLeftOffset = 86;

  // Set appearanceGeneral as a general appearance of the menu
  menu.setAppearance(appearanceGeneral);
  ...
}
```

Note though, that `GEMAppearance` doesn't have default values, so it is necessary to set each property explicitly.

Also it is possible to call `GEMPage::setAppearance()` method on each page individually, passing pointer to `GEMAppearance` object (rather than object itself in case of `GEM::setAppearance()` method).

```cpp
// Create empty GEMAppearance object (its values can be populated later in sketch).
// Note that it should be created in a global scope of the sketch (in order to be passed as a pointer to menu page)
GEMAppearance appearanceSettings;

// Create menu object (its appearance settings will be populated with default values)
GEM menu(glcd);

...

// Later in sketch, e.g. in setupMenu()
void setupMenu() {
  ...
  // Create GEMAppearance object with general values (that will be used for every menu page if not overridden)
  GEMAppearance appearanceGeneral;
  appearanceGeneral.menuPointerType = GEM_POINTER_ROW;
  appearanceGeneral.menuItemsPerScreen = GEM_ITEMS_COUNT_AUTO;
  appearanceGeneral.menuItemHeight = 10;
  appearanceGeneral.menuPageScreenTopOffset = 10;
  appearanceGeneral.menuValuesLeftOffset = 86;

  // Set appearanceGeneral as a general appearance of the menu
  menu.setAppearance(appearanceGeneral); // Note there is no `&` operator when setting general (or global) appearance of the menu

  // Copy values from appearanceGeneral object to appearanceSettings for further customization
  appearanceSettings = appearanceGeneral;
  appearanceSettings.menuValuesLeftOffset = 70;
  menuPageSettings.setAppearance(&appearanceSettings); // Note `&` operator
  ...
}
```

That's basically it. [Readme](https://github.com/Spirik/GEM#gemappearance) and [wiki](https://github.com/Spirik/GEM/wiki/Menu-appearance#gemappearance) may provide additional information.

closed 2024-01-11 by Spirik