Spirik/GEM #59: Simpler create layout way and chain operations

TimonPeng, 2022-08-27:
Inspired by [m5ez](https://github.com/M5ez/M5ez#lets-start-with-text-menus) and Qt, maybe any void functions should can be return instance  for chain operations.

Before:
```c++
GEM_u8g2 menu(u8g2);

GEMPage menuPageMain("Main Menu");

int number = -512;
boolean enablePrint = false;

GEMItem menuItemInt("Number:", number);
GEMItem menuItemBool("Enable print:", enablePrint);

GEMPage menuPageSystemSettings("SYSTEM SETTINGS");
GEMItem menuItemSystemSettings("System Settings", menuPageSystemSettings);

void setup() {
  // ...
  menu.setSplash(logo_width, logo_height, logo_bits);
  menu.hideVersion();
  menu.enableCyrillic();

  menuPageSystemSettings.setParentMenuPage(menuPageMain);

  menuPageMain.addMenuItem(menuItemInt);
  menuPageMain.addMenuItem(menuItemBool);
  menuPageMain.addMenuItem(menuItemSystemSettings);

  menu.setMenuPageCurrent(menuPageMain);
  // ...
}
```

After:
```c++
GEM_u8g2 menu(u8g2)
    .setSplash(logo_width, logo_height, logo_bits)
    .hideVersion()
    .enableCyrillic();

int number = -512;
boolean enablePrint = false;

GEMPage menuPageSystemSettings("SYSTEM SETTINGS")
    .setParentMenuPage(menuPageMain);

GEMPage menuPageMain("Main Menu")
    .addMenuItem("Number:", number)
    .addMenuItem("Enable print:", enablePrint);
    .addMenuItem("System Settings", menuPageSystemSettings);

void setup() {
  // ...
  menu.setMenuPageCurrent(menuPageMain);
  // ...
}
```

TimonPeng, 2022-08-27:
I wrote a demo.

GEM Proxy:
```c++
#ifndef _GEM_PROXY_H
#define _GEM_PROXY_H

#include <Arduino.h>
#include <GEM_u8g2.h>

// https://blog.stratifylabs.dev/device/2020-12-15-Method-Chaining-in-Cpp/
class GEMProxy : public GEM_u8g2 {
 public:
  using GEM_u8g2::GEM_u8g2;
  byte _menuItemHeight;

  GEMProxy& setMenuItemHeight(byte menuItemHeight_) {
    _menuItemHeight = menuItemHeight_;
    return *this;
  };
  GEMProxy& setSplash(byte width, byte height, const unsigned char* image) {
    GEM_u8g2::setSplash(width, height, image);
    return *this;
  };
  GEMProxy& hideVersion() {
    GEM_u8g2::hideVersion();
    return *this;
  };
  GEMProxy& enableCyrillic() {
    GEM_u8g2::enableCyrillic();
    return *this;
  };
};

#endif
```

Usage:
```c++
GEMProxy menu = GEMProxy(u8g2);

void setup() {
  menu.setMenuItemHeight(12)
      .setSplash(logo_width, logo_height, logo_bits)
      .hideVersion()
      .enableCyrillic();
}
```

Spirik, 2022-08-27:
That's a great suggestion, never thought about it! Will definitely look into it.

Spirik, 2023-11-18:
Now available in GEM ver. [1.4.6](https://github.com/Spirik/GEM/releases/tag/1.4.6).

closed 2023-11-18 by Spirik