Spirik/GEM #60: freezing on RP2040

iitztoss, 2022-09-06:
Hi, I have an RP2040-Plus board (which is pin compatible with a Pi Pico).
I'm using the arduino-pico core, ssd1306 OLED, internal 2MB flash
I noticed that when entering a context all works fine, but when I try to exit the program freezes up.
I took the party hard example and changed the u8g2 declaration to
`U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ 5, /* data=*/ 4);   // ESP32 Thing, HW I2C with pin remapping`
and the `u8g2.begin()` part to
`u8g2.begin(/*Select/OK=*/ 26, /*Right/Next=*/ 21, /*Left/Prev=*/ 17, /*Up=*/ 19, /*Down=*/ 22, /*Home/Cancel=*/ 20);`
the rest is the same
Tried to change compiler optimization flags but no success.
also without removing float edit support it won't allow me to edit all other types, just simple checkboxes

Spirik, 2022-09-07:
Hello, @iitztoss !

Unfortunately, never had a chance to work with RP2040 boards yet, however I tested GEM with Arduino Zero and M0 boards (with SAMD chipset that features the same Cortex M0+ core) - they seem to work fine.

Did you try [Blink](https://github.com/Spirik/GEM/wiki/Example-02%3A-Blink) example, it also features context-based action?

As per float issues, may be it has something to do with avr/dtostrf.h support. But [messages](https://forum.arduino.cc/t/why-is-dotostrf-not-declared-for-the-arduino-nano-rp2040-connect/882384) online suggests that it should work fine. Consider manually adding `#include <avr/dtostrf.h>` to your sketch, or commenting this [if-statement](https://github.com/Spirik/GEM/blob/master/src/GEM_u8g2.cpp#L42) in source code (leaving `#include <avr/dtostrf.h>` line intact to force its inclusion regardless of `ARDUINO_ARCH_SAMD` and `ARDUINO_ARCH_SAM` flags).

iitztoss, 2022-09-07:
I just tried that, I can edit values fine, but trying to exit blink v1 (delay-based) doesn't even register the action; trying to exit blink v2 (millis-based) freezes just as before.

Spirik, 2022-09-07:
Can I ask you to add some `Serial.println()` outputs to [blinkMillisContextExit()](https://github.com/Spirik/GEM/blob/master/examples/U8g2/Example-02_Blink/Example-02_Blink.ino#L217) function, that may help us to get an idea at which stage freezing occurs?

iitztoss, 2022-09-07:
Ok, so i did this.
`void blinkMillisContextExit() {
  // Reset variables
  previousMillis = 0;
  labelOn = false;
  Serial.println("labelon");
  
  // Draw menu back on screen and clear context
  menu.reInit();
  Serial.println("reinit");
  menu.drawMenu();
  Serial.println("draw");
  menu.clearContext();
  Serial.println("clear");

  Serial.println("Exit millis based Blink");
}`
I noticed that it does not in fact freeze, but on `reInit()` it takes a minute to perform all tasks, on `drawMenu()` another one, on `clearContext()` it is instant and then nothing is shown on the screen (waited for 5 minutes)
Do you have any idea on how I could debug the library? I tried opening `GEM.cpp` and putting `Serial.println()` all over the functions but it doesn't seem to do anything

Spirik, 2022-09-07:
That's really interesting behavior.

You should try putting `Serial.println()` in GEM_u8g2.cpp since you are using U8g2 version of GEM. Here are links to the corresponding methods: [GEM_u8g2::reInit()](https://github.com/Spirik/GEM/blob/master/src/GEM_u8g2.cpp#L208), [GEM_u8g2::drawMenu()](https://github.com/Spirik/GEM/blob/master/src/GEM_u8g2.cpp#L236), [GEM_u8g2::clearContext()](https://github.com/Spirik/GEM/blob/master/src/GEM_u8g2.cpp#L227).

iitztoss, 2022-09-09:
Right, so I did some tests and it turns out that it takes quite some time to perform each step in reInit, especially in the powerSave function.
Every call is like 10-15 seconds each, and powerSave takes almost a full minute to execute.

I've since halted the development of my program trying to find a way to debug code with a more appropriate IDE, since Arduino 1.x is atrocious.
I suppose there could be a problem with u8g2 or, even worse, the rp2040 core

Spirik, 2022-09-09:
One thing to try is to comment out initDisplay() and setPowerSave() calls from [u8g2::reInit()](https://github.com/Spirik/GEM/blob/master/src/GEM_u8g2.cpp#L208):
```cpp
// _u8g2.initDisplay();
// _u8g2.setPowerSave(0);
````

I wonder if that will make any difference.

As per IDE: I personally use [VS Code](https://code.visualstudio.com/) with [Arduino](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino) extension. Another popular solution for embedded development is [Platformio](https://platformio.org/)  (which is also based on VS Code).

Spirik, 2023-12-29:
Digging up this thread)

Finally got around to test GEM on RP2040 Picoduino board (the one in UNO form-factor). Tested with SH1106 display (which is very much similar to SSD1306) in U8g2 version of GEM with two variations: push-button key detection done with U8g2 (like in original example); and rotary encoder detection done with KeyDetector. Seems to work fine, I was able to enter and exit context with no problem and w/o any abnormal delays.

iitztoss, 2023-12-29:
Hey, thanks for digging up the old thread!
Unfortunately, I reckon I may have abandoned the project a long time ago and I don't even remember what I was working on.
Still, thanks for your dedication to this project, all the best!

Spirik, 2023-12-29:
Got it, thank you for an answer! Closing this issue then for now.

closed 2023-12-29 by Spirik