Spirik/GEM #61: Encoder or/and 4x4_over_i2c_multiplexer

Furbysz, 2022-09-25:
Howdy. First of all, this lib is very nice, seems easy to use and I couldn't find many others that can use SPI 128x64 display.

But back on the topic. I wanted to use it in my project with uses quite a lot of pins rn on its own. Soo reusing 7 just for menu when i have already 16 keys on 4x4 keyboard, thats plugged into i2c multiplexer, soo actually it uses 2 pins of micro controler.
I tought using 2 rotary encoders will do the trick, with they didn't becouse of how they work.

Soo i tought if there is a way of doing it, or easy way to implement those funcionalities directlly. Mebe by using something like functions insead of pins. like instead of pin 6 being home/cancel, calling `GEM_cancel()` would do same.

Thanks in advance.

shezik, 2022-09-26:
This could help.
https://github.com/Spirik/GEM/wiki/How-to#control-menu-from-your-sketch

Spirik, 2022-09-26:
Hi, @Furbysz ! Yes, @shezik is correct, technically you can use whichever input device you have, just register its keypresses (or whatever action) in your own sketch and manually trigger `GEM::registerKeyPress()` method accordingly.

Here is an example of controlling GEM via [Serial](https://github.com/Spirik/GEM/issues/57#issuecomment-1228954728) (roughly inspired by [KeyDetector](https://github.com/Spirik/KeyDetector) library) that may help to get a basic idea of how to write custom control for your menu.

Furbysz, 2022-09-26:
Lokks like clean solution :)

Will take a while to test it tho, code i've wrote doesnt want to work, i wipl post it when i get on my pc.

It combines fastLED, GEM and Keyboard over i2c to probide menu that can be used to select led and change its R, G, and B values.

FastLed and keyboard didnt seem to make problems before, soo i guess its something with gem not working with my setup. I will add mire info when i get on PC.

Other clue can be that im using ESP32 dev module.

Spirik, 2022-09-26:
That is possible. There are [reports](https://github.com/Spirik/GEM/issues/55) of incompatibility with some ESP32 boards: sketch may compile and upload just fine, but issues appear during runtime.

Furbysz, 2022-09-27:
Well, id did not compiled at all. Was screaming at me that it couldnt find **AltSerialGraphicLCD**, after i installed it...
**SoftwareSerial.h** not found. ffs. INSTALL.... used arduino_740163 instad of arduino_979120 whatever those are.
i relocated 740... elsewhere soo it must use 979...
Couldnt find **avr/interrupt.h**

This is a downwards spiral lmao

Spirik, 2022-09-28:
Yes, ESP32 boards do not tolerate AltSerialGraphicLCD library, however you can exclude support for this library prior to compile (by manually editing [config.h](https://github.com/Spirik/GEM/blob/master/src/config.h) file or by setting build flags). See the following Readme sections for details:
- [Compatibility](https://github.com/Spirik/GEM#compatibility)
- [Configuration](https://github.com/Spirik/GEM#configuration)

Furbysz, 2022-10-01:
Seems like it did a trick. It started working. Out of curiosity tho. Can one make menu options programically?

Like when ESP lunches it checks for files on SD card and puts them as options in separate menu. 
Soo when you select in menu "test1.wav" it plays this file.

Spirik, 2022-10-02:
Yes, it should be possible to create menu items in a loop, based on some content (in your case - the count of files on SD card). You can specify the same callback for these menu items, and pass parameter to it to determine which menu item was actually activated (in your case it can be the name of an audio file).

Please see this [post](https://github.com/Spirik/GEM/issues/37#issuecomment-1154203538) with a basic example on how to achieve that (the whole thread was about dynamic menu items creation and with the latest GEM update it became possible).

Spirik, 2023-09-28:
👋  FYI: just updated GEM with [examples](https://github.com/Spirik/GEM/wiki/Example-05%3A-Encoder-U8g2-version) on how to control menu with rotary encoder using latest version of KeyDetector

closed 2023-11-20 by Spirik