Spirik/GEM #65: Floating point

HajmamadAgha, 2022-11-10:
In a page in menus I need to set floating point numbers, my main problem can be defined this way: "after setting the number to a N digit number I can't set the number to a N+1 digit number". How can I resolve this?

Spirik, 2022-11-10:
Hi!

If I understand correctly, you want to extend number of digits after the decimal sign. You have to do this through `GEMItem::setPrecision()` method beforehand (e.g. when creating menu item, before entering an edit mode of this variable).

For more details regarding mantissa and `setPrecision()` method see [Floating-point variables](https://github.com/Spirik/GEM#floating-point-variables) section in Readme and [Create `float` menu item](https://github.com/Spirik/GEM/wiki/How-to#create-float-menu-item) in wiki.

HajmamadAgha, 2022-11-29:
Thanks, Alexander. I think I misinformed my question.
My problem is with the integer part. suppose I set the number to 5.0. After doing that I wouldn't be able to extend that to 50.0 because the factor of ten is removed.

HajmamadAgha, 2022-11-29:
Or if i want to say better, I want to keep the zeros when showing and changing the number

Spirik, 2022-12-07:
Sorry I wasn't able to answer sooner!

I've recorded an example [video](https://www.youtube.com/watch?v=7LdwSz8LAls) showing how you can place decimal point anywhere in the number in edit mode. Basically you're editing a string with limited set of characters, so it is not a problem to change 5.0 -> 50.0 or vice versa, you just may have to edit the whole number to achieve that and that is understandable inconvenience, but is possible.

closed 2023-09-28 by Spirik