Spirik/GEM #68: core 1 panic'ed (loadprohibited) on registerKeyPress

celsoluiz81, 2023-03-05:
Hello,

I'm using GEM in combination with a rotaryEncoder on a ESP32 board. 

I've created a main menu with some options and cyclically call the code below:
` 

if(menu.readyForKey()) {

          byte keyCode = encoder_status;
       
          menu.registerKeyPress(keyCode);
        }
The **encoder_status**  is initiated = GEM_KEY_NONE.

Whenever a change in the encoder value is detected, I calculate the delta and set the value of **encoder_status** accordingly:

- delta > 0 , then **encoder_status** = **GEM_KEY_DOWN**
- delta < 0,  then **encoder_status** = **GEM_KEY_UP**
- button pressed, then **encoder_status** = **GEM_KEY_OK**

Unfortunately, every time **menu.registerKeyPress(keyCode);** is called with an input value other than **GEM_KEY_NONE**, it causes my board to restart with the following trace below. Any tip what could be happening? 

Thank you very much for your time. 

`
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014dc  PS      : 0x00060e30  A0      : 0x800d48e4  A1      : 0x3ffb1e80  
A2      : 0x800d3fd7  A3      : 0x800d3fd3  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x000000aa  A9      : 0x00000040  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x0000ffff  
A14     : 0x0000ffff  A15     : 0x3ffb0060  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x800d3fd7  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400014dc:0x3ffb1e80 0x400d48e1:0x3ffb1e90 0x400d4901:0x3ffb1eb0 0x400d339c:0x3ffb1ed0 0x400d39ad:0x3ffb1ef0 0x400d39ee:0x3ffb1f10 0x400d3f7a:0x3ffb1f30 0x400d3fd4:0x3ffb1f50 0x400d0ac1:0x3ffb1f70 0x400d0773:0x3ffb1f90 0x400d4de1:0x3ffb1fb0 0x40089a2e:0x3ffb1fd0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

`

Spirik, 2023-03-05:
Hello! Thank you for using GEM!

Unfortunately I don't have an ESP32 board on hands at a time to try to replicate the issue. So, let me ask some clarification questions, then=)

1) Which graphics library are you using with GEM?
2) Did you manage to get examples that come with the GEM library working (with KeyDetector or U8g2 based key detection)?
3) Are you sure, that at no given time `registerKeyPress()` is supplied with value that is outside of the valid range, i.e. not from `GEM_KEY_` set of constants?

celsoluiz81, 2023-03-05:
Hi,

Thanks for your prompt reply.

In answer to your questions:

1. Which graphics library are you using with GEM?
_I'm using u8g2_

2. Did you manage to get examples that come with the GEM library working (with KeyDetector or U8g2 based key detection)?
_I wasn't aware these examples applied to me. They seemed to be aimed at key detection, but I am using a rotary encoder. Did I miss something? What I#m doing is that I'm looking at the delta of the encoder value and then calling `registerKeyPress(GEM_KEY_UP)`or `registerKeyPress(GEM_KEY_DOWN)` depending if the delta is positive or negative_

3. Are you sure, that at no given time registerKeyPress() is supplied with value that is outside of the valid range?
_Yes, I'm very sure. The problem happens exactly during the call to `registerKeyPress()`. If I comment it out or replace it with a print call, there is no reset when turning the encoder._

Spirik, 2023-03-05:
Thank you for your answers!

Examples supplied with the library indeed rely on push-buttons to interact with the menu. I was just curious, whether you tried them first before switching to encoder. That would be helpful to know to rule out any possible issues with encoder implementation.

There is another way to test that, though, without the need of building push-buttons circuit: control menu via Serial. [Here is an example](https://github.com/Spirik/GEM/issues/57#issuecomment-1228954728) of such code in a form of a small library (loosely based on KeyDetector logic). That would be a great way to check if an issue is indeed within `registerKeyPress()` method.

One more thing to test out: will sketch crash if `encoder_status` initialized with something other than `GEM_KEY_NONE` (e.g. with `GEM_KEY_CANCEL`)? If problem lies within `registerKeyPress()` then it should crash right after launch/initialization.

celsoluiz81, 2023-03-05:
Hi,

Thanks for the reply. I have seen the example via Serial. What I'm doing is in essence the same: mapping some value to GEM_KEY_UP or GEM_KEY_DOWN. No matter how many ways I spin it, it crashes at the `registerKeyPress()` call.

I tried out your suggestion and initialized `encoder_status` with **GEM_KEY_CANCEL**. Unfortunately, it crashed cyclically and never get passed the splash screen...

celsoluiz81, 2023-03-05:
Maybe another clue:

When I print out the value of the constants `GEM_KEY_UP` and `GEM_KEY_DOWN` I get **84** and **85** respectively. These are the values that are essentially being passed to `registerkeyPress()`. Is this correct?

Spirik, 2023-03-05:
Yes, those values seem about right for U8g2 version of GEM (I map them to U8g2 ones [here](https://github.com/Spirik/GEM/blob/master/src/GEM_u8g2.h#L55)).

Spirik, 2023-03-06:
So far I wasn't able to find anything suspicious in GEM code. Just to make sure, can you please confirm, that calling `registerKeyPress()` _only_ with with `GEM_KEY_NONE` is not causing an error (while _any_ of the values from the list `GEM_KEY_UP`, `GEM_KEY_DOWN`, `GEM_KEY_RIGHT`, `GEM_KEY_LEFT`, `GEM_KEY_CANCEL`, `GEM_KEY_OK` trigger panic)?

And can you please specify exact model of the board you are using? I might have to get it for myself to be able to debug this properly.

Spirik, 2023-03-06:
And if you are using PlatformIO it is possible to specify an option for better debugging, as suggested [here](https://www.reddit.com/r/esp32/comments/wkvyh7/comment/ijr68pu/?utm_source=reddit&utm_medium=web2x&context=3): add `monitor_filters = esp32_exception_decoder` to your `platformio.ini`.

With any luck it may give us a line of source code that ultimately fails.

celsoluiz81, 2023-03-07:
Hi.

Thanks so much for your help with this. Yes, all the values are failing except for `GEM_KEY_NONE`. 
Another fun fact: it seems to only reset when I give `registerKeyPress` input values in the **80s** range. 

For example: since I have limited debugging possibilities, I tried out hard coding `key_code` like this:

` if (menu.readyForKey()) {
  
           key_code=250;
          menu.registerKeyPress(key_code);
        } 
`

I set the value of `key_code` manually and then flashed to see what happens. It will only reset when ` 80 < = key_code < 90`

I'm not sure why. I tried lower values like 0,1 and 2 and also higher values such as 90, 91 and 250. They don't trigger a reset.

Please give me a few days until the weekend. I will install platformio and debug as you suggested.

Spirik, 2023-03-08:
No worries! Thank _you_ for sticking with GEM and helping troubleshoot this thing=)

Since `GEM_KEY_` constants in U8g2 version have values in the range 80 <= key_code <= 85 what you see makes sense: `registerKeyPress()` just ignores other values, so nothing happens.

Actual problems may lay deeper, when GEM tries to perform action tied to a specific key. A little bit of internet searching shows that `loadprohibited` panic may be triggered on ESP32 boards when trying to access some value being out of range (or probably not initialized) or dereference null pointer.

celsoluiz81, 2023-03-14:
Hi,

I installed platformio as you suggested and inserted the line to the .ini like so:

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_extra_dirs = ~/Documents/Arduino/libraries
monitor_filters = esp32_exception_decoder
upload_port = /dev/cu.SLAB_USBtoUART
monitor_port = /dev/cu.SLAB_USBtoUART
monitor_speed = 115200

I reproduced the error after compiling and uploading, however I don't know where the exception is being output. In Arduino IDE I got the exception from the Serial Monitor. Any idea where I should look in platformio? As far as I know I cannot debug without additional hardware.

celsoluiz81, 2023-03-14:
I got it now

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d454c  PS      : 0x00060730  A0      : 0x800d4990  A1      : 0x3ffb21b0  
A2      : 0x3ffc4090  A3      : 0x00000000  A4      : 0x00000014  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x0000000a  A8      : 0x800d454c  A9      : 0x3ffb2190  
A10     : 0x3ffc40f4  A11     : 0x00000001  A12     : 0x0000001e  A13     : 0x0000007e  
A14     : 0x0000001e  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000b  LBEG    : 0x40089aa5  LEND    : 0x40089ab5  LCOUNT  : 0xfffffffd  

Backtrace: 0x400d4549:0x3ffb21b0 0x400d498d:0x3ffb21d0 0x400d49c0:0x3ffb21f0 0x400d4f4a:0x3ffb2210 0x400d4fa8:0x3ffb2230 0x400d2d22:0x3ffb2250 0x400d29ef:0x3ffb2270 0x400d8aad:0x3ffb2290

  #0  0x400d4549:0x3ffb21b0 in U8G2::setDrawColor(unsigned char) at /Users/celsoluiz81/Documents/Arduino/libraries/U8g2/src/U8g2lib.h:218
      (inlined by) GEM_u8g2::drawMenuPointer() at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:447
  #1  0x400d498d:0x3ffb21d0 in GEM_u8g2::drawMenu() at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:242
  #2  0x400d49c0:0x3ffb21f0 in GEM_u8g2::nextMenuItem() at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:479
  #3  0x400d4f4a:0x3ffb2210 in GEM_u8g2::dispatchKeyPress() at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:927
      (inlined by) GEM_u8g2::dispatchKeyPress() at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:864
  #4  0x400d4fa8:0x3ffb2230 in GEM_u8g2::registerKeyPress(unsigned char) at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:861
  #5  0x400d2d22:0x3ffb2250 in WorkstationLcdMain() at src/workstation_lcd.cpp:63
  #6  0x400d29ef:0x3ffb2270 in loop() at src/main.ino:35
  #7  0x400d8aad:0x3ffb2290 in loopTask(void*) at /Users/celsoluiz81/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: 26de181d168ba376

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[  1058][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[  1058][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
Setup done
Initializing motion sensor
[  2707][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (2684) gpio: gpio_set_level(226): GPIO output gpio_num error
Initializing relays
.
Connected to the WiFi network
Local ESP32 IP: 
192.168.178.77

Spirik, 2023-03-15:
Thank you! That's really helpful=)

Issue seems to be tracing back to [U8G2::setDrawColor()](https://github.com/olikraus/U8g2_Arduino/blob/master/src/U8g2lib.h#L218) method, which in turn calls [u8g2_SetDrawColor()](https://github.com/olikraus/U8g2_Arduino/blob/master/src/clib/u8g2_hvline.c#L249), and, presumably, fails when GEM tries to supply the former one with `2` (i.e. [_u8g2.setDrawColor(2)](https://github.com/Spirik/GEM/blob/master/src/GEM_u8g2.cpp#L445)).

Which version of [U8g2](https://github.com/olikraus/U8g2_Arduino) are you using? Support for for value `2` in [setDrawColor()](https://github.com/olikraus/u8g2/wiki/u8g2reference#setdrawcolor) appeared since the version 2.11 (though I am not sure that it would have triggered an error when supplied with invalid value even in earlier versions). I currently run version 2.32.15 (not the lates one).

One thing to try (other than checking and upgrading version of U8g2) is to try to [set up](https://github.com/Spirik/GEM/wiki/Menu-appearance) GEM with `GEM_POINTER_DASH` for the cursor instead of default `GEM_POINTER_ROW` - in this case there will be less calls to `_u8g2.setDrawColor(2)`, and if that is really the case, navigation may work fine (not the edition of variables though, which still relies on `_u8g2.setDrawColor(2)`).

celsoluiz81, 2023-03-19:
Hi,

I tried to setup GEM as you suggested, but I'm still seeing the exceptions. As for the library version I am using 2.33.15. 

When I press the OK button, I get a different exception (Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.), which means:

**InstrFetchProhibited**
_This CPU exception indicates that the CPU could not read an instruction because the address of the instruction does not belong to a valid region in instruction RAM or ROM.

Usually, this means an attempt to call a function pointer, which does not point to valid code. PC (Program Counter) register can be used as an indicator: it will be zero or will contain a garbage value (not 0x4xxxxxxx)._

This could make sense, because I haven't configured any sub menus or options for the menu option I'm trying to enter into.

So this raises the question  about the other exception (LoadProhibited) I'm getting related to `setDrawColor`:

- I'm using a 128x64 LCD display. Monochrome display. Does the setDrawColor method also work with such display?
- Can it be that I have not setup my menu correctly? Perhaps something is missing, causing the exception below to the thrown? I will post my code snippet here later today.

**LoadProhibited, StoreProhibited**
_These CPU exceptions happen when an application attempts to read from or write to an invalid memory location. The address which has been written/read is found in the EXCVADDR register in the register dump. If this address is zero, it usually means that the application has attempted to dereference a NULL pointer. If this address is close to zero, it usually means that the application has attempted to access a member of a structure, but the pointer to the structure is NULL. If this address is something else (garbage value, not in 0x3fxxxxxx - 0x6xxxxxxx range), it likely means that the pointer used to access the data is either not initialized or has been corrupted._

celsoluiz81, 2023-03-19:
Hi,

I fixed some problems with my code and the way I was building the menu.
The exception with `setDrawColor` seems gone now, however I see new exceptions related to the functions `getMenuItemNext` and `dispatchKeyPress` when navigating UP/DOWN and Pressing OK Button respectively (see trace below).

The new exceptions happen also when navigating Up/Down and when pressing OK button as previously reported. 

Below is the code snippets about how I am creating the menu structure. Can you help me check if there are any mistakes?

**This is defined globally:**

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, WORKSTATION_LCD_PIN_E,  WORKSTATION_LCD_PIN_RW, WORKSTATION_LCD_PIN_RS, WORKSTATION_LCD_PIN_RST);

GEMPage menuPageMain("Main Menu");

GEM_u8g2 menu(u8g2,GEM_POINTER_DASH);

These function get called once during init:

`u8g2.begin();
  u8g2.clearBuffer();  menu.init(); WorkstationLcdMainMenu();`

`void WorkstationLcdMainMenu() {
    
    // Add menu items to menu page  
    GEMItem menuItemAmbience("Ambience", WorkstationLcdNetworkMenu);
    menuPageMain.addMenuItem(menuItemAmbience);

    GEMItem menuItemContainer("Container", WorkstationLcdNetworkMenu);
    menuPageMain.addMenuItem(menuItemContainer);

    GEMItem menuItemGarden("Garden", WorkstationLcdNetworkMenu);
    menuPageMain.addMenuItem(menuItemGarden);

    GEMItem menuItemNetwork("Network", WorkstationLcdNetworkMenu);
    menuPageMain.addMenuItem(menuItemNetwork);

    GEMItem menuItemWeather("Weather", WorkstationLcdNetworkMenu);
    menuPageMain.addMenuItem(menuItemWeather);
   
    // Add menu page to menu and set it as current
    menu.setMenuPageCurrent(menuPageMain);
    menu.drawMenu();
}
`
This gets called cyclically :

`   if (menu.readyForKey()) {
            menu.registerKeyPress(joystick_status);
          }  
`

**New Error when navigating:**

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4014d1ab  PS      : 0x00060730  A0      : 0x800d4008  A1      : 0x3ffb2150  
A2      : 0x0000ffff  A3      : 0x3f402a21  A4      : 0x00000400  A5      : 0x3ffc4638  
A6      : 0x3ffb8188  A7      : 0x00000040  A8      : 0x000000fe  A9      : 0x00000007  
A10     : 0x00000000  A11     : 0x00000015  A12     : 0x3f4013a7  A13     : 0x3ffc4140  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0001000b  LBEG    : 0x40089981  LEND    : 0x40089991  LCOUNT  : 0xfffffffd  

Backtrace: 0x4014d1a8:0x3ffb2150 0x400d4005:0x3ffb2170 0x400d4639:0x3ffb2190 0x400d491f:0x3ffb21d0 0x400d4958:0x3ffb21f0 0x400d4ef1:0x3ffb2210 0x400d4f4c:0x3ffb2230 0x400d2e1a:0x3ffb2250 0x400d2adf:0x3ffb2270 0x400d8d95:0x3ffb2290

  #0  0x4014d1a8:0x3ffb2150 in GEMItem::getMenuItemNext() at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEMItem.cpp:1362
  #1  0x400d4005:0x3ffb2170 in GEMPage::getMenuItem(unsigned char, bool) at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEMPage.cpp:87 (discriminator 3)
  #2  0x400d4639:0x3ffb2190 in GEM_u8g2::printMenuItems() at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:310
  #3  0x400d491f:0x3ffb21d0 in GEM_u8g2::drawMenu() at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:241
  #4  0x400d4958:0x3ffb21f0 in GEM_u8g2::nextMenuItem() at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:479
  #5  0x400d4ef1:0x3ffb2210 in GEM_u8g2::dispatchKeyPress() at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:925
      (inlined by) GEM_u8g2::dispatchKeyPress() at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:864
  #6  0x400d4f4c:0x3ffb2230 in GEM_u8g2::registerKeyPress(unsigned char) at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:861
  #7  0x400d2e1a:0x3ffb2250 in WorkstationLcdMain() at src/workstation_lcd.cpp:69
  #8  0x400d2adf:0x3ffb2270 in loop() at src/main.ino:37
  #9  0x400d8d95:0x3ffb2290 in loopTask(void*) at /Users/celsoluiz81/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

**Error when pressing OK button:**

Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00010001  PS      : 0x00060730  A0      : 0x800d4ed6  A1      : 0x3ffb21f0  
A2      : 0x00010001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00060b20  A7      : 0x00000001  A8      : 0x800d4ce4  A9      : 0x000000ff  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0xb33fffff  
A14     : 0x007b833c  A15     : 0x003fffff  SAR     : 0x0000001b  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00010000  LBEG    : 0x40089981  LEND    : 0x40089991  LCOUNT  : 0xfffffffa  

Backtrace: 0x0000fffe:0x3ffb21f0 0x400d4ed3:0x3ffb2210 0x400d4f4c:0x3ffb2230 0x400d2e1a:0x3ffb2250 0x400d2adf:0x3ffb2270 0x400d8d95:0x3ffb2290

  #0  0x0000fffe:0x3ffb21f0 in ?? ??:0
  #1  0x400d4ed3:0x3ffb2210 in GEM_u8g2::dispatchKeyPress() at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:942
      (inlined by) GEM_u8g2::dispatchKeyPress() at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:864
  #2  0x400d4f4c:0x3ffb2230 in GEM_u8g2::registerKeyPress(unsigned char) at /Users/celsoluiz81/Documents/Arduino/libraries/GEM/src/GEM_u8g2.cpp:861
  #3  0x400d2e1a:0x3ffb2250 in WorkstationLcdMain() at src/workstation_lcd.cpp:69
  #4  0x400d2adf:0x3ffb2270 in loop() at src/main.ino:37
  #5  0x400d8d95:0x3ffb2290 in loopTask(void*) at /Users/celsoluiz81/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

celsoluiz81, 2023-03-19:
I got it working. Thanks!

As I suspected, it was my code not being setup correctly. I had declared the items below in a setup function and not globally as per your example. 

GEMItem menuItemAmbience("Ambience", WorkstationLcdNetworkMenu);
GEMItem menuItemContainer("Container", WorkstationLcdNetworkMenu);
GEMItem menuItemGarden("Garden", WorkstationLcdNetworkMenu);
GEMItem menuItemNetwork("Network", WorkstationLcdNetworkMenu);
GEMItem menuItemWeather("Weather", WorkstationLcdNetworkMenu);

Thanks for the support.

closed 2023-03-19 by celsoluiz81

Spirik, 2023-03-20:
Awesome! Glad that everything worked out for you!