Spirik/GEM #79: Не отображается значение в пункте меню на дисплее HX1230

gausstechno, 2023-11-27:
Дисплей с контроллером HX1230 (или же STE2007, типа Nokia 1220)
ESP32-WROOM-32 (с гравировкой QR кода) на собственной плате
GEM 1.4.6
U8g2 2.34.22
Arduino IDE 2.2.1

Не могу понять, почему не отображаются значение в пунктах меню?
В названии пункта числа отображаются, а в значении - нет

Загружаю примеры с кириллицей - ничего, не с кириллицей - тоже ничего, в чём может быть причина?
Делаю список опций - не работает, вывожу просто значение с помощью GEMItem digit("Цифра", cyfra); и всё так же нет значения cyfra - int, пробовал int, byte, через GEMSelect - тоже ничего.

[image]

Spirik, 2023-11-27:
@gausstechno Привет!

На такой комбинации железа лично не тестил, увы. Но первое, что приходит в голову: для начала попробовать отключить поддержку float'ов в [config.h](https://github.com/Spirik/GEM/blob/master/src/config.h) , как описано вот [тут](https://github.com/Spirik/GEM#floating-point-variables) (даже если непосредственно в проекте float'ов нет, косвенно может быть связано). И заодно лишние библиотеки (AltSerialGraphicLCD и Adafruit GFX), как описано [ниже](https://github.com/Spirik/GEM#configuration). Т.е. просто закомментить строки:
```cpp
// #include "config/enable-glcd.h"

// #include "config/enable-adafruit-gfx.h"

// #include "config/support-float-edit.h"
```

Spirik, 2023-11-27:
От IDE не должно зависеть, у самого такая же=) С ESP32 периодически пишут о проблемах, но ко мне отладочная плата на нём приедет чуть позже. Но и не факт, что в этом случае дело в нём.

Есть ещё такая гипотеза. Судя по скриншоту, места для значения переменной на экране немного, судя по положению чекбокса всего 1 символ. Может быть просто значение левого смещения подобрано таким образом, что алгоритм решил, что не уместится и не стал рисовать. Думаю, можно попробовать при инициализации `GEM_u8g2` передать другое значение `menuValuesLeftOffset_` (последний аргумент в конструкторе). 

Из характеристик чипа дисплея, что я нагуглил, выходит что он всего 96x68? При смещении в 86px слева это действительно могло стать причиной. Нужно попробовать уменьшить его, скажем, до 62.

gausstechno, 2023-11-27:
Йоууу, это сработало! Я подозревал, что ему не хватает места, но не совсем понимал, куда копать.

U8G2_HX1230_96X68_F_3W_SW_SPI u8g2();

При этой инициализации надо бы тогда эти параметры по умолчанию добавить, было бы классно)

Spirik, 2023-11-27:
Надо будет подумать, как облегчить задание параметров кастомизации меню. Например, позволять задавать их не только большой цепочкой аргументов через конструктор, но и по-отдельности через методы, как было предложено [тут](https://github.com/Spirik/GEM/issues/58). Подумаю в этом направлении.

Рад, что удалось разобраться!=)

closed 2023-11-27 by Spirik

Spirik, 2023-12-28:
Привет! Чуть апну тему.

В версии [1.5.0](https://github.com/Spirik/GEM/releases/tag/1.5.0) появилась возможность задавать стили меню индивидуально для каждой страницы: это может быть полезным для небольших экранов (например, когда требуется на страничке настроек задать побольше места для названий пунктов в случае, если сами значения представляют собой чекбоксы и большой отступ для них не нужен).

Так же теперь можно задавать значения параметров внешнего вида поимённо через объект типа `GEMAppearance`.

Примеры использования есть в [Readme](https://github.com/Spirik/GEM#gemappearance) и [wiki](https://github.com/Spirik/GEM/wiki/Menu-appearance#gemappearance).