Spirik/GEM #84: Switch button behavior for 3-buttons navigations

albydnc, 2024-03-11:
Hello,

I have a display with only 3 buttons (up, down, enter). I would like to use the library to create menus, but to do so, I have to switch the key behaviour when the menu is in editing mode or in navigation mode.
Is it possible to have the feedback of the menu state to switch behaviour?

Thanks

Spirik, 2024-03-12:
Hello, @albydnc !

Yes, it is possible! Boolean property, which holds current mode of the menu (edition of variable or navigation through menu items), is called [`_editValueMode`](https://github.com/Spirik/GEM/blob/master/src/GEM_u8g2.h#L188) . It is `protected` (since GEM ver. [1.5.2](https://github.com/Spirik/GEM/releases/tag/1.5.2)), which makes it possible to access this property  from within user-defined derived (inherited) class of GEM. Based on the example provided in [How to](https://github.com/Spirik/GEM/wiki/How-to#inherit-classes) section of wiki, it will look something like this:

```cpp
// Define GEMProxy class which inherits from GEM_u8g2
class GEMProxy : public GEM_u8g2 {
  public:
    using GEM_u8g2::GEM_u8g2;

    // Get _editValueMode
    bool isInEditMode() {
      return _editValueMode;
    }
};

...

// Use GEMProxy class to create menu here
GEMProxy menu(u8g2);

...

// Access user-defined method to get value of internal _editValueMode property
menu.isInEditMode();
```

Another possible solution to operation of the GEM with only 3 buttons that I may suggest is to detect combinations of key presses (i.e. simultaneous presses of two buttons), available in latest version of [KeyDetector](https://github.com/Spirik/KeyDetector?tab=readme-ov-file#simultaneous-presses). That way it is possible to implement mapping of this 3 buttons to 6 key codes GEM expects, e.g. like so (given buttons <kbd>A</kbd>, <kbd>B</kbd>, <kbd>C</kbd>:
* <kbd>A</kbd>: `GEM_KEY_UP`
* <kbd>B</kbd>: `GEM_KEY_DOWN`
* <kbd>A</kbd> + <kbd>C</kbd>: `GEM_KEY_RIGHT`
* <kbd>B</kbd> + <kbd>C</kbd>: `GEM_KEY_LEFT`
* <kbd>C</kbd> short press: `GEM_KEY_OK`
* <kbd>C</kbd> long press: `GEM_KEY_CANCEL`

Though you would have to carefully write your own key detection routine to correctly measure time intervals (delays) and key combinations. Some of this approach can be seen in Rotary Encoder [examples](https://github.com/Spirik/GEM/wiki/Example-05%3A-Encoder) (which uses short and long-presses of push-button and combinations of button presses with the rotation of the knob).

Hope that helps=)

shezik, 2024-03-13:
This method is oddly familiar 😛

closed 2024-03-20 by albydnc

garudaonekh, 2024-12-18:
> Hello, @albydnc !
> 
> Yes, it is possible! Boolean property, which holds current mode of the menu (edition of variable or navigation through menu items), is called [`_editValueMode`](https://github.com/Spirik/GEM/blob/master/src/GEM_u8g2.h#L188) . It is `protected` (since GEM ver. [1.5.2](https://github.com/Spirik/GEM/releases/tag/1.5.2)), which makes it possible to access this property from within user-defined derived (inherited) class of GEM. Based on the example provided in [How to](https://github.com/Spirik/GEM/wiki/How-to#inherit-classes) section of wiki, it will look something like this:
> 
> ```c++
> // Define GEMProxy class which inherits from GEM_u8g2
> class GEMProxy : public GEM_u8g2 {
>   public:
>     using GEM_u8g2::GEM_u8g2;
> 
>     // Get _editValueMode
>     bool isInEditMode() {
>       return _editValueMode;
>     }
> };
> 
> ...
> 
> // Use GEMProxy class to create menu here
> GEMProxy menu(u8g2);
> 
> ...
> 
> // Access user-defined method to get value of internal _editValueMode property
> menu.isInEditMode();
> ```
> 
> Another possible solution to operation of the GEM with only 3 buttons that I may suggest is to detect combinations of key presses (i.e. simultaneous presses of two buttons), available in latest version of [KeyDetector](https://github.com/Spirik/KeyDetector?tab=readme-ov-file#simultaneous-presses). That way it is possible to implement mapping of this 3 buttons to 6 key codes GEM expects, e.g. like so (given buttons A, B, C:
> 
> * A: `GEM_KEY_UP`
> * B: `GEM_KEY_DOWN`
> * A + C: `GEM_KEY_RIGHT`
> * B + C: `GEM_KEY_LEFT`
> * C short press: `GEM_KEY_OK`
> * C long press: `GEM_KEY_CANCEL`
> 
> Though you would have to carefully write your own key detection routine to correctly measure time intervals (delays) and key combinations. Some of this approach can be seen in Rotary Encoder [examples](https://github.com/Spirik/GEM/wiki/Example-05%3A-Encoder) (which uses short and long-presses of push-button and combinations of button presses with the rotation of the knob).
> 
> Hope that helps=)

The approach of combining A+C to navigate Right is not convenient. 

Can we just customize the rotary encoder for 3-button? 

Please point me to the places to update. First time using your library.

Thanks;

Spirik, 2024-12-18:
Hello @garudaonekh !

It is possible to use GEM with rotary encoder, see examples [here](https://github.com/Spirik/GEM/wiki/Example-05%3A-Encoder) and description of demo stand [here](https://github.com/Spirik/GEM/wiki/Example-05%3A-Encoder).

It is, of course, also possible to use push buttons to navigate through menu. However, you should implement your own mapping of the button presses (or combination of presses) to key codes that GEM understands (`GEM_KEY_UP`, `GEM_KEY_RIGHT`, `GEM_KEY_DOWN`, `GEM_KEY_LEFT`, `GEM_KEY_CANCEL`, `GEM_KEY_OK`). See [Control menu from your sketch](https://github.com/Spirik/GEM/wiki/How-to#control-menu-from-your-sketch) section in wiki for description of each code.

garudaonekh, 2024-12-19:
> Control menu from your sketch

Thanks for your quick response. 

I think it's better to convert rotary encoder to 3 buttons than combining the six buttons to three buttons.

Is it a feasible solution?

garudaonekh, 2024-12-19:
thanks for you quick response.

I want to eliminate KEY_LEFT and RIGHT. But I see the issue is with updating number and string beyond 1 character.
Is it easy to have 0 to 999 up/down number update and support long press of up/down to increase the number scrolling speed?

My end goal is to eliminate key combination in order to create LEFT/RIGHT event

Spirik, 2024-12-19:
Indeed, lack of `GEM_KEY_RIGHT` and `GEM_KEY_LEFT` interaction will significantly restrict an ability to edit variables, however it will be still possible to navigate menu and interact with checkboxes, selects and range spinners (which is a form of select essentially).

In case of numbers I'd suggest to use [GEMSpinner](https://github.com/Spirik/GEM/wiki/How-to#create-int-value-spinner) (see [Wokwi](https://wokwi.com/projects/410889496248755201) simulation). Or use [GEMSelect](https://github.com/Spirik/GEM/wiki/How-to#create-int-option-select) when more verbose labeling of values is required.

As for adjusting "scrolling speed", I think it should be somewhat possible by calling the `menu.registerKeyPress()` method several times in a row (that will make select jump multiple values at once). You can try to differentiate between short and long presses the same (or similar) way it is done for `GEM_KEY_OK`/`GEM_KEY_CANCEL` button in [rotary encoder](https://github.com/Spirik/GEM/wiki/Test-bench-with-rotary-encoder-for-U8g2-version#use) examples.

All-in-all the following scheme should be feasible enough:

| Buttons                  | Key identifier   | Action                 |
|--------------------------|------------------|------------------------|
| Button A                | `GEM_KEY_DOWN`   | Navigate down through the menu items list, select next option in select |
| Button B               | `GEM_KEY_UP`     | Navigate up through the menu items list, select previous option in select |
| Button C Short press              | `GEM_KEY_OK`     | Toggle bool menu item, enter edit mode of the associated non-bool variable, exit edit mode with saving the variable, execute code associated with button |
| Button C Long press               | `GEM_KEY_CANCEL` | Navigate to the previous (parent) menu page, exit edit mode without saving the variable, exit context loop if allowed within context's settings |

garudaonekh, 2024-12-20:
Thanks for all the detail and point me to the correct direction.  I am customizing it and a lot to learn.

I already switched from GEM_adafruit_gfx(Adafruit_ST7735) to TFT_espi instead due to font rendering of non-ascii code.