Spirik/GEM #93: Add std::function callbacks feature

vortigont, 2024-04-17:
Hi there!
Would be really nice to have functional callbacks in menu items. I.e. like [here](https://github.com/h2zero/NimBLE-Arduino/issues/122). This is required to use capturing lambda functions or class instance members for menu callbacks.
Thanks!

Spirik, 2024-04-17:
Hi, @vortigont!

Seems like a great suggestion! I will look into it. I believe I had similar ideas when added support for current version of callbacks, but did not proceed with the implementation (probably was not sure about wide enough Arduino support: quick search gives [mixed](https://community.platformio.org/t/arduino-c-11-or-c-14-support-in-platform-io/15683/3) results on `<functional>` support across different platforms). Also a little worried that using `#include <functional>` may lead to (potentially not insignificant) increase in library compiled size, but that needs to be verified.

vortigont, 2024-04-18:
yes, for some platforms STL might not available. Actually most of the popular menu libs I could find ([u8g2](https://github.com/olikraus/u8g2/issues/2258), [ArduinoMenu](https://github.com/neu-rah/ArduinoMenu), + some others) are not friendly to lambdas or object member callbacks, though being implemented as cpp classes. Portability is nice to have, but today is 2024 with many 32 bit MCUs around where you do not need to struggle for every byte of mem.
Having some contemporary alternative would be really nice.
Desperately thinking about writing some implementation of my own :)

Spirik, 2024-04-18:
Technically, I can make inclusion `#include <functional>` optional (and update methods declaration in GEMItem.h accordingly) by implementing new [configuration](https://github.com/Spirik/GEM?tab=readme-ov-file#configuration) flag (e.g. `GEM_ENABLE_FUNCTIONAL_CALLBACK`).

Alternatively, in current version it is possible to enable [Advanced mode](https://github.com/Spirik/GEM?tab=readme-ov-file#advanced-mode) and override declaration of `GEMItem` constructors in your own sketch (with the means to add support for std::function suitable for platform of your choice).

Spirik, 2024-04-19:
And just to make sure you've considered this option (and for posterity), there is another possible workaround available in current version of GEM. As a part of a callback it is possible to pass user-defined value - it can be pointer to an object you would like to associate with the menu item. In that way while callback itself is a static function, you will have access to a desired object (and e.g. call that object's specific method or in any other way manipulate it).

See [this section](https://github.com/Spirik/GEM/wiki/How-to#add-menu-item-callback-with-callback-arguments) in wiki for details and [GEMCallbackData](https://github.com/Spirik/GEM?tab=readme-ov-file#gemcallbackdata) reference in Readme.

vortigont, 2024-04-19:
With `(void*)` pointer parameter I think it is possible to use some typecasting and lambda magic to access object instance member. I will look into it, tnx for the hint!

BTW, when working with wowki and encoder how could I move to the next char when creating strings?
I was able to make only one letter elements [here](https://wokwi.com/projects/385198727246499841)

Spirik, 2024-04-19:
> BTW, when working with wowki and encoder how could I move to the next char when creating strings?
> I was able to make only one letter elements [here](https://wokwi.com/projects/385198727246499841)

Use simultaneous press of the knob and rotation, it will result in `GEM_KEY_LEFT` and `GEM_KEY_RIGHT` signals (see description of [demo stand](https://github.com/Spirik/GEM/wiki/Test-bench-with-rotary-encoder-for-U8g2-version#use) in wiki). In wokwi you can press <kbd>Space</kbd> (button press) in combination with arrow keys (rotation) to navigate horizontally (do not forget to focus encoder element first).

closed 2024-05-10 by Spirik

vortigont, 2024-06-29:
@Spirik just in case you might be interested I came up with own implementation of menu-like lib which is based on abstract classes and functional callbacks - [MuiPlusPlus](https://github.com/vortigont/MuiPlusPlus)

Spirik, 2024-06-30:
> @Spirik just in case you might be interested I came up with own implementation of menu-like lib which is based on abstract classes and functional callbacks - [MuiPlusPlus](https://github.com/vortigont/MuiPlusPlus)

Looks like a great option for more performant MCUs! Congrats on starting to publicly release your builds!