Spirik/GEM #96: Compilation errors with LilyGo TTGO T-Display

gadgetmies, 2024-07-02:
I'm trying to build the Example-01_Basic.ino for the LilyGo TTGO T-Display. When compiling I get a lot of errors such as the following:

```
~/Documents/Arduino/libraries/GEM/src/GEMItem.cpp: In constructor 'GEMItem::GEMItem(const char*, byte&, GEMSelect&, void (*)(GEMCallbackData))':
~/Documents/Arduino/libraries/GEM/src/GEMItem.cpp:96:38: error: either all initializer clauses should be designated or none of them should be
   96 |   , callbackData{ .pMenuItem = this, { 0 } }
      |                                      ^
```

The failing compile command is (likely) as follows:
```
~/Library/Arduino15/packages/esp32/tools/esp-x32/2302/bin/xtensa-esp32-elf-g++ -MMD -c @~/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/flags/cpp_flags -w -Os -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_LILYGO_T_DISPLAY -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"LILYGO_T_DISPLAY\"" "-DARDUINO_VARIANT=\"lilygo_t_display\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"macosx\"" "-DARDUINO_FQBN=\"esp32:esp32:lilygo_t_display:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=16M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 @~/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/flags/defines -I~/Documents/Projects/Code/nutrient_mixer -iprefix ~/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/include/ @~/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/flags/includes -I~/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32/qio_qspi/include -I~/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/cores/esp32 -I~/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/variants/lilygo_t_display -I~/Documents/Arduino/libraries/KeyDetector/src -I~/Documents/Arduino/libraries/Ai_Esp32_Rotary_Encoder/src -I~/Documents/Arduino/libraries/Adafruit_ST7735_and_ST7789_Library -I~/Documents/Arduino/libraries/Adafruit_GFX_Library -I~/Documents/Arduino/libraries/Adafruit_BusIO -I~/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/libraries/Wire/src -I~/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/libraries/SPI/src -I~/Documents/Arduino/libraries/GEM/src -I~/Documents/Arduino/libraries/U8g2/src @/private/var/folders/mt/pp6hn_fx1j5b67vzkvxk217h0000gn/T/arduino/sketches/23D5DF54B931F69BDFF8D10FC97C386F/build_opt.h @/private/var/folders/mt/pp6hn_fx1j5b67vzkvxk217h0000gn/T/arduino/sketches/23D5DF54B931F69BDFF8D10FC97C386F/file_opts ~/Documents/Arduino/libraries/GEM/src/GEMItem.cpp -o /private/var/folders/mt/pp6hn_fx1j5b67vzkvxk217h0000gn/T/arduino/sketches/23D5DF54B931F69BDFF8D10FC97C386F/libraries/GEM/GEMItem.cpp.o
```

The error seems like it could be caused by the use of a different compiler or a different version of the compiler. Has anyone else had similar issues or know what could cause such errors?

Spirik, 2024-07-03:
@gadgetmies Hi there!

Let's investigate! I'd like to clarify a few things first. Is [this](https://www.lilygo.cc/products/lilygo%C2%AE-ttgo-t-display-1-14-inch-lcd-esp32-control-board) the board in question? Since display is color-capable (and ST7789V based, according to specs), I assume you are using Adafruit GFX version of GEM (and probably the latest 1.5.2 one). Is this correct? Does Adafruit GFX examples compile and run on that board?

gadgetmies, 2024-07-03:
The board is probably a clone of the one you linked to, but seems identical: https://www.aliexpress.com/item/1005005970553639.html. I am using Adafruit GFX version of GEM and would expect the version to be the latest as I installed it using the Arduino IDE library manager just yesterday. I will check the versions and whether the Adafruit GFX examples compile later today.

Spirik, 2024-07-03:
Seems like example compiles successfully for ST7789 based display with Adafruit_ST7789.h library using ESP32 Dev Module settings (which is a viable solution for this board according to "Technology Support" section on the site of [original](https://www.lilygo.cc/products/lilygo%C2%AE-ttgo-t-display-1-14-inch-lcd-esp32-control-board) board).

It also supports initialization via `TFT_eSPI` but I didn't test that one (but googled some possible issues with that [here](https://forum.arduino.cc/t/lilygo-ttgo-t-display-not-working-with-tft-espi/1185743)).

ST7789-specific modifications of the sketch marked with red here:

[image]

Spirik, 2024-07-03:
Also it seems that recent versions of compiler have more strict rules regarding initializer lists and designated initializers (notably [C++20](https://mariusbancila.ro/blog/2020/02/27/c20-designated-initializers/)) . That may be the reason you are seeing this particular error.

AFAIK, there is no straightforward way to change compiler version in Arduino IDE. And most community members [don't recommend](https://forum.arduino.cc/t/c-version-compiler/1247642) it either. Though it might be not a version per se, but some configuration settings of default Arduino C++11 compiler?

Spirik, 2024-07-06:
Error messages now addressed and fixed in release [1.5.3](https://github.com/Spirik/GEM/releases/tag/1.5.3).

The issue was due to more strict regulations of [designated initializers](https://mariusbancila.ro/blog/2020/02/27/c20-designated-initializers/) in recent compilers starting with C++20. And recent update of ESP32 core version 3.0 (that comes with boards update via Arduino IDE) uses C++23.

gadgetmies, 2024-07-06:
> Error messages now addressed and fixed in release [1.5.3](https://github.com/Spirik/GEM/releases/tag/1.5.3).

Awesome! Thank you so much @Spirik!

closed 2024-07-06 by Spirik