Spirik/GEM #97: Problem with ESP32 S3

motniemtin, 2024-07-04:
```
/libraries/GEM/src/GEMItem.cpp: In constructor 'GEMItem::GEMItem(const char*, byte&, GEMSelect&, void (*)(GEMCallbackData))':
/libraries/GEM/src/GEMItem.cpp:96:38: error: either all initializer clauses should be designated or none of them should be
   96 |   , callbackData{ .pMenuItem = this, { 0 } }
```

My demo Project https://wokwi.com/projects/402474219113695233

[image]

Spirik, 2024-07-04:
I was able to reproduce the issue after updating ESP32 boards via Arduino IDE. Will work on the fix next.

motniemtin, 2024-07-05:
Hope you update soon

Spirik, 2024-07-06:
Now fixed in release [1.5.3](https://github.com/Spirik/GEM/releases/tag/1.5.3).

The issue was due to more strict regulations of [designated initializers](https://mariusbancila.ro/blog/2020/02/27/c20-designated-initializers/) in recent compilers starting with C++20. And recent update of ESP32 core version 3.0 (that comes with boards update via Arduino IDE) uses C++23.

Wokwi demo compiles successfully now (you may need to remove and re-add GEM library for new version to be fetched):
[image]

motniemtin, 2024-07-06:
Thank you so much !!!

closed 2024-07-06 by Spirik