Spirik/GEM #104: Waveshare 1.5inch OLED Module (B) SH1107

PDE-Paul, 2024-09-03:
Hi,
I am wanting to use the Waveshare 1.5inch OLED Module in I2C mode with an address of 0x3D, I cannot get anything to display on the screen. I have made sure I am calling display.display(); after menu.drawMenu(); as per another forum post I found a callback can be used so I implemented this, still nothing is displayed on the screen.

I can get text and images to display correctly using the Adafruit_GFX.h and Adafruit_SH110X.h without GEM included, so I know my hardware is working. I just cannot get GEM to initialize the display, is there a trick to this? I am using a custom ESP32 board, GPIO setup below in pins_arduino.h 

Here is my pins_arduino.h
`#ifndef Pins_Arduino_h
#define Pins_Arduino_h

#include <stdint.h>

// UART
static const uint8_t TX = 1;
static const uint8_t RX = 3;

#define RXD2 4
#define TXD2 16

//I2C
static const uint8_t SDA = 5;
static const uint8_t SCL = 17;

//SPI
//static const uint8_t SS    = 5;
static const uint8_t MOSI = 2;
static const uint8_t MISO = 15;
static const uint8_t SCK  = 14;

//Analog
static const uint8_t A0 = 36;
static const uint8_t A3 = 39;
static const uint8_t A4 = 32;
static const uint8_t A5 = 33;

//Input
static const uint8_t Inp1 = 36;
static const uint8_t Inp2 = 39;
static const uint8_t Inp3 = 34; 
static const uint8_t Inp4 = 35;

//Output
static const uint8_t Out1 = 32;
static const uint8_t Out2 = 33;
static const uint8_t Out3 = 12;
static const uint8_t Out4 = 13;

//Ethernet LAN8720A
#define ETH_PHY_TYPE    ETH_PHY_LAN8720
#define ETH_PHY_ADDR    1
#define ETH_PHY_MDC     23
#define ETH_PHY_MDIO    18
#define ETH_PHY_POWER   -1
#define ETH_CLK_MODE  	ETH_CLOCK_GPIO0_OUT
//#define ETH_PHY_POWER -1
// TX_EN 	= 21
// TD0 		= 19
// TXD1 	= 22
// RXD0 	= 25
// RXD1 	= 26
// CRS_DV 	= 27

//SD Card
static const uint8_t SD_Present = 39;
static const uint8_t SS    		= 33;
#endif /* Pins_Arduino_h */
`

Here is my sketch
`#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SH110X.h>
#include <GEM_adafruit_gfx.h>
//#include <KeyDetector.h>

// Pins the buttons are connected to
//const byte downPin = 2;
//const byte leftPin = 3;
//const byte rightPin = 4;
//const byte upPin = 5;
//const byte cancelPin = 6;
//const byte okPin = 7;

// Array of Key objects that will link GEM key identifiers with dedicated pins
//Key keys[] = {{GEM_KEY_UP, upPin}, {GEM_KEY_RIGHT, rightPin}, {GEM_KEY_DOWN, downPin}, {GEM_KEY_LEFT, leftPin}, {GEM_KEY_CANCEL, cancelPin}, {GEM_KEY_OK, okPin}};

// Create KeyDetector object
//KeyDetector myKeyDetector(keys, sizeof(keys)/sizeof(Key));
// To account for switch bounce effect of the buttons (if occur) you may want to specify debounceDelay
// as the third argument to KeyDetector constructor:
// KeyDetector myKeyDetector(keys, sizeof(keys)/sizeof(Key), /* debounceDelay= */ 10);

#define SCREEN_WIDTH 128  // OLED display width, in pixels
#define SCREEN_HEIGHT 128 // OLED display height, in pixels
#define OLED_RESET -1     // can set an oled reset pin if desired
Adafruit_SH1107 display = Adafruit_SH1107(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET, 1000000, 100000);

// Create variables that will be editable through the menu and assign them initial values
int number = 10;
bool enablePrint = false;

// Create two menu item objects of class GEMItem, linked to number and enablePrint variables 
GEMItem menuItemInt("Number:", number);
GEMItem menuItemBool("Enable print:", enablePrint);

// Create menu button that will trigger printData() function. It will print value of our number variable
// to Serial monitor if enablePrint is true. We will write (define) this function later. However, we should
// forward-declare it in order to pass to GEMItem constructor
void printData(); // Forward declaration
GEMItem menuItemButton("Print", printData);

// Create menu page object of class GEMPage. Menu page holds menu items (GEMItem) and represents menu level.
// Menu can have multiple menu pages (linked to each other) with multiple menu items each
GEMPage menuPageMain("Main Menu");

// Create menu object of class GEM_adafruit_gfx. Supply its constructor with reference to tft object we created earlier
GEM_adafruit_gfx menu(display, GEM_POINTER_ROW, GEM_ITEMS_COUNT_AUTO);
// Which is equivalent to the following call (you can adjust parameters to better fit your screen if necessary):
// GEM_adafruit_gfx menu(tft, /* menuPointerType= */ GEM_POINTER_ROW, /* menuItemsPerScreen= */ GEM_ITEMS_COUNT_AUTO, /* menuItemHeight= */ 10, /* menuPageScreenTopOffset= */ 10, /* menuValuesLeftOffset= */ 86);

void drawMenuCallback() {
  display.display();
}

void setupMenu() {
  // Add menu items to menu page
  menuPageMain.addMenuItem(menuItemInt);
  menuPageMain.addMenuItem(menuItemBool);
  menuPageMain.addMenuItem(menuItemButton);

  // Add menu page to menu and set it as current
  menu.setMenuPageCurrent(menuPageMain);
}

void setup() {
  // Push-buttons pin modes
//  pinMode(downPin, INPUT);
//  pinMode(leftPin, INPUT);
//  pinMode(rightPin, INPUT);
//  pinMode(upPin, INPUT);
//  pinMode(cancelPin, INPUT);
//  pinMode(okPin, INPUT);

  // Serial communications setup
  Serial.begin(115200);

  display.begin(0x3D);         // Address 0x3D default

  // Menu init, setup and draw
  menu.setDrawMenuCallback(drawMenuCallback);
  menu.init();
  setupMenu();
  menu.drawMenu();
}

void loop() {
  delay(1000);
  Serial.println("Testing");
}

void printData() {
  // If enablePrint flag is set to true (checkbox on screen is checked)...
//  if (enablePrint) {
    // ...print the number to Serial
//    Serial.print("Number is: ");
//    Serial.println(number);
//  } else {
//    Serial.println("Printing is disabled, sorry:(");
//  }
}
`

I have disabled any key input, does this need to be enabled to display something on the screen?

Any assistance would be greatly appreciated
Kind Regards
Paul Brink

Spirik, 2024-09-03:
Welcome back, @PDE-Paul =)

First thing to check. In Adafruit GFX it may be required to set foreground and background colors of an OLED display explicitly (even for monochrome displays). Please, check if this helps. (The following snippet of `setup()` routine is from [Example-05_Encoder](https://github.com/Spirik/GEM/blob/master/examples/AdafruitGFX/Example-05_Encoder/Example-05_Encoder.ino) which uses an [OLED screen](https://github.com/Spirik/GEM/wiki/Test-bench-with-rotary-encoder-for-Adafruit-GFX-version)).

```cpp
  // Serial communication setup
  Serial.begin(115200);

  delay(250);  // Wait for the OLED to power up
  display.begin(0x3D, true);  // Address 0x3C default
  
  display.display();
  delay(2000);

  // Clear the buffer
  display.clearDisplay();

  // Explicitly set correct colors for monochrome OLED screen
  menu.setForegroundColor(SH110X_WHITE);
  menu.setBackgroundColor(SH110X_BLACK);
  
  // Menu init, setup and draw
  menu.init();
  setupMenu();
  menu.drawMenu();

  display.display();

  Serial.println("Initialized");
```

If this won't work, then check Serial output for "Initialized" message (which is added at the end of `setup()` routine above).

Spirik, 2024-09-03:
Also, does your initialization code come from Adafruit example? Seems that your display is equipped with SSD1327 controller, and the one that imported is `#include <Adafruit_SH110X.h>` . I know it is totally possible that they are compatible enough (like SH1106 and SH1306 for example), but just to be sure.

PDE-Paul, 2024-09-03:
Thank you for the guidance... its working... here is my altered setup()

`void setup() {
  // Push-buttons pin modes
//  pinMode(downPin, INPUT);
//  pinMode(leftPin, INPUT);
//  pinMode(rightPin, INPUT);
//  pinMode(upPin, INPUT);
//  pinMode(cancelPin, INPUT);
//  pinMode(okPin, INPUT);

  // Serial communications setup
  Serial.begin(115200);
  delay(250);
  display.begin(0x3D);         // Address 0x3D default
  display.display();
  delay(2000);
  display.clearDisplay();
  display.setTextColor(SH110X_WHITE); // Draw white text

// Explicitly set correct colors for monochrome OLED screen
  menu.setForegroundColor(SH110X_WHITE);
  menu.setBackgroundColor(SH110X_BLACK);

  // Menu init, setup and draw
  menu.setDrawMenuCallback(drawMenuCallback);
  menu.init();
  setupMenu();
  menu.drawMenu();
}
`

Now I can continue getting Navigation buttons working, you can close this issue off.

Again much appreciated the quick help.

Kind Regards
Paul Brink

closed 2024-09-03 by Spirik