Spirik/GEM #105: Alignment (Question)

PDE-Paul, 2024-09-11:
Hi,

I am looking to align my menu items from the center of the screen, is this possible? if so how do I do this in code. Should I use GEMAppearance? 

I have previous used this routine and it displays center alignment no matter how wide my screen is, I am using Adafruit GFX in the this routine

`void drawCentreString(const String &buf, int x, int y) {

    int16_t x1, y1;

    uint16_t w, h;

    display.getTextBounds(buf, 0, 0, &x1, &y1, &w, &h); //calc width of new string

    display.setCursor((SCREEN_WIDTH - w) / 2, y);

    display.print(buf);

  }
`

How would I push the setCursor location into GEM?

Regards
Paul

Spirik, 2024-09-12:
Hi, Paul!

Internally GEM considers all available screen space and is drawn from the top left corner of the display. You can [adjust](https://github.com/Spirik/GEM/wiki/Menu-appearance) height and number of menu items (thus saving some space at the bottom of the screen), height of the title bar (thus adding some additional space at the top of the screen), but menu items will always take all available screen width.

PDE-Paul, 2024-09-12:
Thank you, I will use GEMAppearance to tweak my alignment then.

closed 2024-09-15 by Spirik