Spirik/GEM #110: Move to edit mode of a menu item

garudaonekh, 2024-12-31:
Hi,
I am trying to implement a step by step setting. i.e after user set Hour, show Minute spinner. and so on. I based on your todo list example.

Current status: I can show Minute spinner and move focus to it after user set Hour. 

But I can't figure out how to go to edit mode of Minute. I try to use menu.registerKeyPress(KEY_OK) to mimick user click but it crash.

Spirik, 2024-12-31:
Hi! Can your provide snippet of your code?

Are you adding minutes spinner dynamically via `addMenuItem()` or just show item that was previously hidden (via `show()`)? Dynamic memory management can be tricky (and not all that recommended on Arduino boards and MCUs in general), so it is better to avoid it whenever possible.

How are you moving focus to Minute spinner (there is one small example in wiki which can be found [here](https://github.com/Spirik/GEM/wiki/How-to#inherit-classes), but I wonder, whether you are using another solution, e.g. calling `registerKeyPress()` to navigate to it)?

garudaonekh, 2024-12-31:
When Hour menu item is set, Minute menu item is shown and move focus to it. But I can't find a way to go to its edit mode directly unless user click on OK.

```

GEMItem menuItemProgramHour("em:agebIk",pHour, spinnerHour, pHourCallback, GEM_VAL_BYTE);

void pHourCallback(GEMCallbackData callbackData){
  menuItemProgramMinute.hide(false);
  ((GEMPageProxy*)menu.getCurrentMenuPage())->setCurrentItemNum(2);
  menu.drawMenu();
}
```

Spirik, 2025-01-01:
I was able to reproduce an issue and will prepare fix. Thank you for bringing this to my attention!

The issue itself lies in order at which user-defined callback is called and inner mechanics of exiting an edit mode. `callbackAction` firing before inner method `exitEditValue()` may lead to not-intuitive results, e.g. when author of the sketch tries to call `registerKeyPress()`, menu state at that point will operate with assumption that is is still in edit mode (`_editValueMode == true`). So callback fires when associated with menu item variable is saved but before exit from edit mode.

garudaonekh, 2025-01-02:
> I was able to reproduce an issue and will prepare fix. Thank you for bringing this to my attention!
>
> The issue itself lies in order at which user-defined callback is called and inner mechanics of [...]

Rather than sending GEM_KEY_OK, i call menuItemSelect() directly by moving it to public function also doesn't work(nothing happen).

Spirik, 2025-01-02:
Solution with `registerKeyPress()` should work fine after the fix is implemented. I also intend to make `setCurrentItemNum()` method public, so you won't have to override `GEMPage` class in order to access it.

However, there may be slight difference in callback code between U8g2 and Adafruit GFX versions of GEM (due to inner workings). Which version of GEM do you use?

garudaonekh, 2025-01-02:
I use TFT_espi. I change your adafruit_GFX tft to TFT_espi instead.

Spirik, 2025-01-02:
Does your display have buffer (e.g. SH110X series), so you have to call `display.display()` after each `drawMenu()` call explicitly in order to send updates to the screen and force actual redraw? Such as the case with [Todo list](https://github.com/Spirik/GEM/blob/master/examples/AdafruitGFX/Example-06_Todo-List/Example-06_Todo-List.ino) example. Or additional call is not needed (e.g. ST7735 based displays), as in the case with other [examples](https://github.com/Spirik/GEM/blob/master/examples/AdafruitGFX/Example-01_Basic/Example-01_Basic.ino)?

garudaonekh, 2025-01-02:
> Does your display have buffer (e.g. SH110X series), so you have to call `display.display()` after each `drawMenu()` call explicitly in order to send updates to the screen and force actual redraw? [...]

Migrating from adafruitGFX ST7735 to TFT_espi is very smooth. both library use virtually all the same method name. The only difference is  TFT_espi use tft.begin() while adafruit_gfx use tft.initR(INITR_BLACKTAB)

 The reason I switch to tft_espi is font rendering issue.

garudaonekh, 2025-01-02:
The work around for  menuItemSelect() to work is to set the flag and call it from loop function().

Spirik, 2025-01-02:
The flag workaround will probably work for `registerKeyPress()` version as well (set flag in your callback then call `registerKeyPress()` from within `loop()`).

garudaonekh, 2025-01-02:
> The flag workaround will probably work for `registerKeyPress()` version as well (set flag in your callback then call `registerKeyPress()` from within `loop()`).

yes, because registerKeyPress(GEM_KEY_OK) will call menuItemSelect().

closed 2025-01-02 by garudaonekh

Spirik, 2025-01-04:
Should be resolved with the [1.6.1](https://github.com/Spirik/GEM/releases/tag/1.6.1) release.

Here is an example of function, that can be called from within user-defined callback to move focus to the next menu item upon saving (and optionally simulating `GEM_KEY_OK` button press):

```cpp
void selectNextMenuItem(bool pressOK = false) {
  // Get current menu item index
  GEMPage* currentMenuPage = menu.getCurrentMenuPage();
  byte menuItemIndex = currentMenuPage->getCurrentMenuItemIndex();
  // Set pointer to the next menu item
  currentMenuPage->setCurrentMenuItemIndex(menuItemIndex + 1);
  // menu.drawMenu(); // Uncomment for Adafruit GFX version for correct redraw (no display.display() call needed here even for buffer-equipped displays)
  if (pressOK) {
    // Simulate GEM_KEY_OK press on the next menu item
    menu.registerKeyPress(GEM_KEY_OK);
  }
}
```

Note that method for setting focus to the menu item by index `GEMPage::setCurrentMenuItemIndex()` is now made public, so no need to write custom overrides through GEMPage inheritance to achieve that.

Two functional examples were prepared to demonstrate how it is possible to create step-by-step wizard-like experience for the user:

* [test_79_gem-u8g2_sh1106_callback-wizard.ino](https://gist.github.com/Spirik/aa840bf011852db25266c5519e694687) - using U8g2 version of GEM. "GEM - Callback Wizard" version of this example (using U8g2 version of GEM) available on [Wokwi](https://wokwi.com/projects/419142050690399233).
* [test_80_gem-agfx_sh1106_callback-wizard.ino](https://gist.github.com/Spirik/05081aa4de7f0a47c9f511f13c50829c) - using Adafruit GFX version of GEM.