Spirik/GEM #118: Ability to hide the top page title

thijstriemstra, 2025-03-06:
By the example, I had to change 

```
GEMPage menuPageMain("Main menu");
```

to 

```
GEMPage menuPageMain("");
```

in order to hide the title above the main menu. I couldn't find an api to hide it any other way. Is there a better way, and if not, I would suggest a feature to hide it. Currently there's no constructor without a label name afaik.

Spirik, 2025-03-07:
I will add support for overloaded call to `GEMPage menuPageMain();` (that will be essentially equivalent to `GEMPage menuPageMain("");`). It shouldn't affect compiled size of the sketch, so it's totally fine to add.

Thank you for suggestion!

thijstriemstra, 2025-03-07:
nice, thanks!

thijstriemstra, 2025-03-10:
fyi, when trying to override `drawTitle`using advanced mode I get the error that `_menuPageCurrent->title` is protected:

```c++
class GEMProxy: public GEM_adafruit_gfx
{
  public:
    using GEM_adafruit_gfx::GEM_adafruit_gfx;

    void drawTitleBar() override {
      _agfx.setFont(_fontFamilies.small);
      _agfx.setTextWrap(true);
      _agfx.setTextColor(_menuForegroundColor);
      _agfx.setCursor(5 * _textSize, _menuItemFont[1].baselineOffset * _textSize + 1);
      _agfx.print(_menuPageCurrent->title);
      _agfx.setTextWrap(false);
      _agfx.setFont(getMenuItemFontSize() ? _fontFamilies.small : _fontFamilies.big);
    }
};
```

so I had to make `const char* title;` public.

Spirik, 2025-03-14:
Support for `GEMPage menuPageMain` call to constructor (without the need to submit an empty string to create menu page without a title, i.e. `GEMPage menuPageMain("")`) available in release [1.6.4](https://github.com/Spirik/GEM/releases/tag/1.6.4).

closed 2025-03-16 by Spirik