Spirik/GEM #119: Ability to hide spinner up/down sprites

thijstriemstra, 2025-03-07:
I'd like to hide the spinner up/down sprites because they're too small to be understood on 280x240 display and it's "valuable real-estate" I'd rather like to use for values. It also provides more room from the label left of the value.

Spirik, 2025-03-07:
Will look into it in details later.

thijstriemstra, 2025-03-07:
Ideally I'd like to right align the values where the sprites are now.

thijstriemstra, 2025-03-07:
i also noticed the sprites aren't hidden when text overlaps them:

[image]

thijstriemstra, 2025-03-08:
> Also noticed that offset of the sprite from the right side of the screen on your latest screenshot is kinda weird: should not be that far from the edge (more like on your previous screenshots).

yeah this is because the display doesn't properly fit my  3d printed case, it's overlapped on left and missing space on right. This is why I would like to adjust it: #120

Spirik, 2025-03-08:
It's not the distance to the edge of the case that caught my attention, it's the distance to the edge of the display itself. I mean, at these photos you've shared distance from the sprite to the edge of the screen is quite different (less on the first one and more on the second one), judging by the extent of the focused menu item pointer. Is it because you've already adjusted source code to try to move placement of the sprite?

[image]

thijstriemstra, 2025-03-08:
Oh I see.

> . Is it because you've already adjusted source code to try to move placement of the sprite?

No. This is vanilla.

thijstriemstra, 2025-03-08:
Here are the fonts I'm using. it's PiRoboto and public domain anyway.

[fonts.zip](https://github.com/user-attachments/files/19146069/fonts.zip)

Using

```c++
// menu
menu.invertKeysDuringEdit(true);
menu.setSplashDelay(0);
// Set custom fonts and specify width, height and baselineOffset of characters
menu.setFontBig(&Piboto_Regular12pt7b, 8, 12, 12).setFontSmall(&Piboto_Bold12pt7b, 8, 12, 12);
menu.setTextSize(1);
menu.setSpriteSize(2);

menu.init();
setupMenu();
menu.drawMenu();
```

thijstriemstra, 2025-03-08:
that overlap issue also seems to have fixed itself somehow:

 [image]

Spirik, 2025-03-08:
Interesting 🤔. Please, let me know, if this occurs again - really looks like a possible bug. Meanwhile, I will checkout this custom font.

thijstriemstra, 2025-03-08:
@Spirik I think it happened when I was testing the `menu.setSpriteSize()` method and set it to `menu.setSpriteSize(3)` or even `menu.setSpriteSize(10)`.. Not sure though.

Spirik, 2025-03-09:
Was able to reproduce and fix. Will publish with the next release (alongside https://github.com/Spirik/GEM/issues/118 and https://github.com/Spirik/GEM/pull/117).

Solution for hiding select and spinner arrows is in the works as well (not sure yet, whether it will be workaround or proper implementation).

Spirik, 2025-03-10:
I decided that for the time being quick and dirty hack will suffice, especially given that you've already familiarized yourself with method overrides in Advanced mode=)

So we will override `GEM_adafruit_gfx::drawSprite()` method. One of its arguments (namely `sprite`) contains object of the class `Sprite` that should be drawn on the screen (alongside its coordinates and color). We just compare its values to the default sprite that GEM provides (we don't have to compare all of the bytes, just some unique ones). And if match is found, we won't draw anything.

```cpp
class GEMProxy: public GEM_adafruit_gfx
{
  public:
    using GEM_adafruit_gfx::GEM_adafruit_gfx;

    void drawSprite(int16_t x, int16_t y, const Splash sprite[], uint16_t color) override {
      byte variant = _spriteSize - 1;
      // Compare bytes to the default selectArrows sprite
      if (sprite[variant].image[2] != 0x70 && sprite[variant].image[6] != 0x1e) {
        _agfx.drawBitmap(x, y, sprite[variant].image, sprite[variant].width, sprite[variant].height, color);
      }
    }
};
```

The more proper solution, as I imagine it, will require some ways to allow override of individual sprites. One way to achieve this is with introduction of callback function that will be called in `GEM_adafruit_gfx::drawSprite()` before sprite is drawn. Author of the sketch will be able to determine what sprite is about to be drawn and act accordingly: override it with own sprite or simply hide it.

That will of course require some amount of development on my part (including unification with U8g2 version of GEM). So I will come back to it some time in the future.

thijstriemstra, 2025-03-10:
> I decided that for the time being quick and dirty hack will suffice, especially given that you've already familiarized yourself with method overrides in Advanced mode=)

Thanks, looks like that worked.

Spirik, 2025-03-14:
> Was able to reproduce and fix. Will publish with the next release (alongside [#118](https://github.com/Spirik/GEM/issues/118) and [#117](https://github.com/Spirik/GEM/pull/117)).

Fix available in release [1.6.4](https://github.com/Spirik/GEM/releases/tag/1.6.4).

closed 2025-03-16 by Spirik