Spirik/GEM #120: appearance: menuPageScreenLeftOffset

thijstriemstra, 2025-03-07:
My 3d-printed case didn't perfectly fit my screen and I'd like to add some padding on the left.

[image]

Spirik, 2025-03-07:
I am afraid that there is no straightforward way to do this in current version of GEM: that may require rather extensive changes to layout logic currently implemented.

You may try for yourself the following hack, though: find in [GEM_adafruit_gfx.h](https://github.com/Spirik/GEM/blob/master/src/GEM_adafruit_gfx.cpp) and change every occurrence of `_agfx.setCursor(5` to `_agfx.setCursor(10`. That will increase left offset from 5px to 10px. However all of the other math won't be affected, so long menu item titles may overflow to the space where values are printed.

thijstriemstra, 2025-03-07:
Thanks for the feedback. My second question related to this was following but I don't think that'll be supported judging by your feedback:

I have a custom "frame"-shape drawn on the screen and would like to place the menu in the center, making it look retro. I suppose this isn't possible, drawing the menu on top of adafruit's gfx canvasses?

Spirik, 2025-03-07:
I think it can be done to some extent, e.g. with the use of [drawMenu() callback](https://github.com/Spirik/GEM/wiki/How-to#add-drawmenu-callback) to draw something on the screen after menu is drawn. That should work best for adding some custom graphics above the menu (where menu title usually is) and below the menu. As for the sides of the screen: that indeed may be challenging, because menu is drawn full-width, from the left border of the screen to the right border with no offset settings to properly adjust that behavior.

But, as usually the case, there are some hacks to try out:) For example you may want to start menu item titles with the space character (i.e. " Enable print" instead of "Enable print"): that will add offset to the left of the menu equal to the width of the space character.

The right side is more challenging, because icons are drawn there. So you would have to adjust source code (to remove icons) or draw something directly on top of them (not sure if this will cause some noticeable flickering).

So all-in-all I'd suggest you to try and draw your frame on top of the menu in `drawMenu()` callback. May be that will produce some decent results after all.

closed 2025-03-16 by Spirik