Spirik/GEM #121: menu.setSpriteSize not working

thijstriemstra, 2025-03-07:
When setting `menu.setSpriteSize(2);` I see the icon becoming bigger but it seems it won't go bigger than that, setting it to a value > 2 doesn't seem to have any effect. It means it's really hard to see in my project with custom fonts:

[image]

Spirik, 2025-03-07:
That is correct, [scale factor](https://github.com/Spirik/GEM/wiki/Menu-appearance#changing-sprite-magnification-adafruit-gfx-version-only) only scales sprites up to 2x times. That is expected behavior: sprites stored in source code in [Bitmap](https://learn.adafruit.com/adafruit-gfx-graphics-library/graphics-primitives#bitmaps-2002806) format and only two versions are defined. You will have to prepare new sprites and adjust corresponding variables in [source code](https://github.com/Spirik/GEM/blob/master/src/GEM_adafruit_gfx.cpp#L132) directly.

thijstriemstra, 2025-03-07:
thanks for the feedback. I think this should be documented (2 is the max).

Spirik, 2025-03-07:
> I think this should be documented (2 is the max). 

There is note on that in [Reference](https://github.com/Spirik/GEM?tab=readme-ov-file#reference) section in Readme (`Sprites (i.e. various menu icons) will be scaled maximum up to two times regardless of the value`) and in [wiki](https://github.com/Spirik/GEM/wiki/Menu-appearance#changing-sprite-magnification-adafruit-gfx-version-only) (`Sprites can be scaled up to 2 times maximum`).

> Is it possible to use word Yes/No instead of the icon with boolean fields?

The easiest way to achieve that (without modifying source code) is to use select menu item with two options labeled `Yes` and `No`:
```cpp
byte invert = 1;
SelectOptionByte selectInvertOptions[] = {{"Yes", 1}, {"No", 0}};
GEMSelect selectInvert(sizeof(selectInvertOptions)/sizeof(SelectOptionByte), selectInvertOptions);
```

The other solution is to enable [Advanced mode](https://github.com/Spirik/GEM?tab=readme-ov-file#advanced-mode) and [override](https://github.com/Spirik/GEM/wiki/How-to#override-methods-in-advanced-mode) GEM methods where checkbox sprites [are](https://github.com/Spirik/GEM/blob/master/src/GEM_adafruit_gfx.cpp#L453) [drawn](https://github.com/Spirik/GEM/blob/master/src/GEM_adafruit_gfx.cpp#L777). Or to edit [those sprites](https://github.com/Spirik/GEM/blob/master/src/GEM_adafruit_gfx.cpp#L155) directly in source code.

closed 2025-03-08 by thijstriemstra

thijstriemstra, 2025-03-09:
thanks, that boolean workaround worked although i'd still like to hide these spinners.

[image]