Spirik/GEM #122: Add eeprom example

thijstriemstra, 2025-03-09:
Storing settings and retreving them using eeprom and/or esp32 settings classes is something a lot of users will find useful.

Spirik, 2025-03-09:
Do you mean storing values of editable variables (menu items)?

Generally you prepare set of variables in your sketch for menu items to edit (as usual); initialize them with values read from EEPROM (or other storage) at the start of the sketch (in `setup()`) or with default values (if storage is empty); and finally write updated values back to the storage in [menu item callback](https://github.com/Spirik/GEM/wiki/How-to#add-menu-item-callback-basic).

I may add simple example in  [Additional examples](https://github.com/Spirik/GEM/wiki/How-to#additional-examples) section in wiki.

Spirik, 2025-03-11:
Done. Modification of Basic example using EEPROM to store data is added to [Addition examples](https://github.com/Spirik/GEM/wiki/How-to#additional-examples) section in wiki.

[Gist](https://gist.github.com/Spirik/31e3d32076658470290e5317e2eed859) and [Wokwi](https://wokwi.com/projects/425151134380135425) simulation (using U8g2 version of GEM) is available.