Spirik/GEM #125: Callback on value change

thijstriemstra, 2025-03-10:
I have a menu with 3 spinners:

- brightness (0 - 255)
- background (White or Black)
- Text (White, Black, Green, Red, etc)

When the user changes these values I want to update associated feature immediately so the user can see the visual changes (e.g. text color of menu changes, backlight brightness updates, etc). Currently the user has to change value, press OK and color will be updated. This works very cumbersome and unintuitive.

Same would apply to a volume control, on every update the user will hear a tone that changes volume depending on the value. When the user is happy, OK is pressed and value is stored.

I tried to find this by hacking `drawEditValueSelect` code:

 ```c++
void drawEditValueSelect() {
      GEMItem* menuItemTmp = _menuPageCurrent->getCurrentMenuItem();
      drawEditValueCursor();
      _agfx.setTextColor(_menuBackgroundColor);
      
      int pointerPosition = getCurrentItemTopOffset(false);
      byte yText = pointerPosition + getMenuItemInsetOffset() + _menuItemFont[getMenuItemFontSize()].baselineOffset * _textSize;
      _agfx.setCursor(getCurrentAppearance()->menuValuesLeftOffset, yText);
      
      switch (menuItemTmp->linkedType) {
        case GEM_VAL_SELECT:
          {
            GEMSelect* select = menuItemTmp->select;
            Serial.print("String Value: ");
            Serial.println(select->getOptionNameByIndex(_valueSelectNum));
            printMenuItemValue(select->getOptionNameByIndex(_valueSelectNum));
          }
          break;
        #ifdef GEM_SUPPORT_SPINNER
        case GEM_VAL_SPINNER:
          {
            char valueStringTmp[GEM_STR_LEN];
            GEMSpinner* spinner = menuItemTmp->spinner;
            GEMSpinnerValue valueTmp = spinner->getOptionNameByIndex(menuItemTmp->linkedVariable, _valueSelectNum);
            switch (spinner->getType()) {
              case GEM_VAL_BYTE:
                itoa(valueTmp.valByte, valueStringTmp, 10);
                break;
              case GEM_VAL_INTEGER:
                itoa(valueTmp.valInt, valueStringTmp, 10);
                break;
              #ifdef GEM_SUPPORT_FLOAT_EDIT
              case GEM_VAL_FLOAT:
                dtostrf(valueTmp.valFloat, menuItemTmp->precision + 1, menuItemTmp->precision, valueStringTmp);
                break;
              case GEM_VAL_DOUBLE:
                dtostrf(valueTmp.valDouble, menuItemTmp->precision + 1, menuItemTmp->precision, valueStringTmp);
                break;
              #endif
            }
            Serial.print("Number Value: ");
            Serial.println(valueStringTmp);
            printMenuItemValue(valueStringTmp);
          }
          break;
        #endif
      }

      //drawSprite(_agfx.width() - 7 * _spriteSize, getCurrentItemTopOffset(true, true), selectArrows, _menuBackgroundColor);
      _agfx.setTextColor(_menuForegroundColor);
    }
```

The override didn't work though (protected member errors) so I had to directly modify the code.

And then it prints:

```
String Value: White
String Value: Black
Number Value: 233
Number Value: 232
```

Which is almost what I need ( I don't want the string "White" but the associated color code with it:

```c++
SelectOptionInt backgroundColors[] = {
  {"Black", ST77XX_BLACK}, {"White", ST77XX_WHITE}
};
```

And then it would have to be called back in order to act on it and change text color etc.

Spirik, 2025-03-11:
There are several ways you can deal with protected fields.

* **First**, as shown in #118 , is to check if there is public getter or setter method available, that will give you access to the desired property. In case of accessing protected field `GEMPage::title` (as was the case in the mentioned issue), it is possible to change the following call:
  ```cpp
  _agfx.print(_menuPageCurrent->title);
  ```
  to
  ```cpp
  _agfx.print(_menuPageCurrent->getTitle());
  ```

* **Second**, if you willing to edit some source code, there is an alternative way to what you did here. No need to change `protected` access specifier of a field to `public`. You can add your proxy class as a friend to the class you need to get access to:
  [image]
That way you will be able to access protected fields of the `GEMPage` from the methods of your `GEMProxy` class directly.

* **Third**, probably slightly cumbersome way, works without the need to modify source code files directly. Create proxy class of the class fields of which you want to access (say `GEMPageProxy`). Change access specifier of the field (here let's stick with the example of `GEMPage::title` field) to public:
  ```cpp
  class GEMPageProxy : public GEMPage {
    public:
      using GEMPage::GEMPage;
      using GEMPage::title;  // Changing access specifier to make this field public
  };
  ```
  Next, cast `GEMPage` class to `GEMPageProxy` by  changing your call to private field of the original `GEMPage` class:
  ```cpp
  _agfx.print(_menuPageCurrent->title);
  ```
  to
  ```cpp
  _agfx.print(((GEMPageProxy*)_menuPageCurrent)->title);
  ```
* **Fourh**, the same as previous, but instead of changing access specifier of the field in our proxy class, we just make it friends with `GEMProxy`:
  ```cpp
  class GEMPageProxy : public GEMPage {
    friend class GEMProxy;  // Making friends
    public:
      using GEMPage::GEMPage;
  };
  ```
  Next, as in previous one, cast `GEMPage` class to `GEMPageProxy` by  changing your call to private field of the original `GEMPage` class:
  ```cpp
  _agfx.print(_menuPageCurrent->title);
  ```
  to
  ```cpp
  _agfx.print(((GEMPageProxy*)_menuPageCurrent)->title);
  ```
  That way we'll get access not only to `GEMPage::title` field, but to all of the other protected fields as well.

thijstriemstra, 2025-03-17:
I went with the 2nd option and added `GEMProxy` as friend class to GEMSpinner, GEMItem, GEMSelect. Any chance you could make this possible in the library without me having to modify the code? Modifying private code is one thing, but modifying source code to add a callback.. Yeah, not ideal.

I was also not able to access the `selectArrows` bits (that I'm not even using!) so I also redefined those by copying:

```c++
#define selectArrows_width  6
#define selectArrows_height 8
static const uint8_t selectArrows_bits [] PROGMEM = {
  0x00, 0x20, 0x70, 0x00, 0x70, 0x20, 0x00, 0x00
};
#define selectArrows_width_scaled  12
#define selectArrows_height_scaled 16
static const uint8_t selectArrows_bits_scaled [] PROGMEM = {
  0x00, 0x00, 0x00, 0x00, 0x0c, 0x00, 0x1e, 0x00, 0x3f, 0x00, 0x3f, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x3f, 0x00, 0x3f, 0x00, 0x1e, 0x00, 0x0c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};
const Splash selectArrows[] = {
  {selectArrows_width, selectArrows_height, selectArrows_bits},
  {selectArrows_width_scaled, selectArrows_height_scaled, selectArrows_bits_scaled}
};
```

Spirik, 2025-03-17:
> Any chance you could make this possible in the library without me having to modify the code?

I am considering it, but there are a lot of possible implications to deal with. So I won't be making my decision just yet.

* Should `onCancel()` (exit without saving) callback be added as well? Because if author of the sketch uses `onValueChange()` callback to dynamically preview changes, how would they know if the selected value was not saved (and instead discarded and rolled back to previous one)?
* How values should be passed to the callback? Should it be `void*` pointer, so that it will be author's responsibility to cast it to the proper data type? Or should it be collection of overloaded methods to set callback for each and every possible supported data type variation?
* Should it be called for editable menu items (such as numbers and `char[]` strings)? What the argument should be then: whole new intermediate value or just the single character?

And probably there are other less obvious things as well.

Other ways, in which desirable behavior can be implemented, should be also considered in more details.

* Say, for example, what if instead `onValueChange()` callback there is a callback, that will be called on each button press, and then state of the GEM will be inspected? Author of the sketch can then check if menu currently is in edit mode (`::isEditMode()` method), then get access to the current menu item (`::getCurrentMenuPage()`->`GEMPage::getCurrentMenuItem()`), and access property of GEM that stores intermediate value of the variable: for editable variables it is called `_valueString` and currently is protected, for selects and spinners index of currently selected item is stored in `_valueSelectNum`, which is also protected. There is, arguably, no need to implement callback at all, because if it is presumably called on each button press, it is possible to implement it directly in `loop()`, where key detection is done. Then all is left to do is to add public methods for accessing `_valueString` and `_valueSelectNum` (or it is possible to access them right away with custom methods in Advanced Mode).

> I was also not able to access the `selectArrows` bits (that I'm not even using!) so I also redefined those by copying:

That's because currently sprites are not part of the class, they stored in constants (was implemented that way initially with space considerations in mind). And that's another story I would like to deal with in some way in the future, when I come up with methods for overriding of the default sprites (relevant to #119 as well).

thijstriemstra, 2025-03-17:
thanks for the feedback. I now have the following override:

```c++
void drawEditValueSelect() override {
      GEMItem* menuItemTmp = _menuPageCurrent->getCurrentMenuItem();
      drawEditValueCursor();
      _agfx.setTextColor(_menuBackgroundColor);

      int pointerPosition = getCurrentItemTopOffset(false);
      byte yText = pointerPosition + getMenuItemInsetOffset() + _menuItemFont[getMenuItemFontSize()].baselineOffset * _textSize;
      _agfx.setCursor(getCurrentAppearance()->menuValuesLeftOffset, yText);

      switch (menuItemTmp->getLinkedType()) {
        case GEM_VAL_SELECT:
          {
            GEMSelect* select = menuItemTmp->select;
            printMenuItemValue(select->getOptionNameByIndex(_valueSelectNum), -1 * calculateSpriteOverlap(selectArrows));

            dispatcher.dispatch(
              MenuEvent::VALUE_UPDATE, select->getType(), select->getOptionNameByIndex(_valueSelectNum)
            );
          }
          break;
        #ifdef GEM_SUPPORT_SPINNER
        case GEM_VAL_SPINNER:
          {
            char valueStringTmp[GEM_STR_LEN];
            GEMSpinner* spinner = menuItemTmp->spinner;
            GEMSpinnerValue valueTmp = spinner->getOptionNameByIndex(menuItemTmp->linkedVariable, _valueSelectNum);
            switch (spinner->getType()) {
              case GEM_VAL_BYTE:
                itoa(valueTmp.valByte, valueStringTmp, 10);
                break;
              case GEM_VAL_INTEGER:
                itoa(valueTmp.valInt, valueStringTmp, 10);
                break;
              #ifdef GEM_SUPPORT_FLOAT_EDIT
              case GEM_VAL_FLOAT:
                dtostrf(valueTmp.valFloat, menuItemTmp->precision + 1, menuItemTmp->precision, valueStringTmp);
                break;
              case GEM_VAL_DOUBLE:
                dtostrf(valueTmp.valDouble, menuItemTmp->precision + 1, menuItemTmp->precision, valueStringTmp);
                break;
              #endif
            }
            printMenuItemValue(valueStringTmp, -1 * calculateSpriteOverlap(selectArrows));

            // notify listeners
            dispatcher.dispatch(
              MenuEvent::VALUE_UPDATE, spinner->getType(), valueStringTmp
            );
          }
          break;
        #endif
      }

      drawSprite(_agfx.width() - 7 * _spriteSize, getCurrentItemTopOffset(true, true), selectArrows, _menuBackgroundColor);
      _agfx.setTextColor(_menuForegroundColor);
    }
```

I realized that instead of making a custom event listener, and re-inventing all the callback logic, it might be simpler to just call the GEM-defined callback for the item. Unfortunatly I cannot find a way to trigger the callback. I tried adding:

```c++
if (menuItemTmp->callbackAction != nullptr) {
  if (menuItemTmp->callbackWithArgs) {
    menuItemTmp->callbackActionArg(menuItemTmp->callbackData);
  } else {
    menuItemTmp->callbackAction();
  }
}
```

but the callback is never fired. Any idea?

Spirik, 2025-03-17:
Consider this version as well. I've expanded on the thought from my latest post. There is no need in Advanced Mode in this one: no overrides, but some proxying to make protected fields and method accessible from the sketch.

* [Gist](https://gist.github.com/Spirik/c6a6f318a5cbf311bb212d3e558798aa) of AGFX version;
* [Gist](https://gist.github.com/Spirik/9abc3e0347fbf45cc659b388458f2324) of U8g2 version;
* [Wokwi simulation](https://wokwi.com/projects/385197253571415041) of U8g2 version.

It is quite trivial to get `_valueSelectNum`, which stores the index of currently selected item of the select or spinner. But in case of spinners you would have to parse it into actual value. You can implement your own logic for that, but I decided to call some of the inner methods of the `GEMSpinner` for that (so I had to proxy them as well).

> but the `validateBrightness` callback is never fired. Any idea?

Hm, not sure, at the first glance looks fine to me. I assume upon saving the variable callback is firing as it should?

Spirik, 2025-03-17:
Oh, I think I might have an idea on what is going on there. You are probably reading variable value in your `validateBrightness` callback, is that correct? But that variable won't be updated until menu item is saved. Only then `valueStringTmp` would be copied to an actual variable replacing its value.

So that might explain why calling `menuItemTmp->callbackAction` does nothing. But I am not sure about your `dispatcher.dispatch()` method - seems like you submit `valueStringTmp` there as an argument. Or does it work as expected?